metrodroid/metrodroid #27: Improve Bilhete Único detection

micolous, 2016-12-28:
Bilhete Único currently looks for `bcdefghi` at the tail end of Sector 0 Block 0 on Mifare Classic cards.

This is actually associated with the card vendor that they use, specifically the [Fudan Microelectronics FM11RF08](http://www.seeedstudio.com/document/FM11RF08(English).pdf) (a Mifare Classic clone).

This picks up many false positives with other agencies and cards which use the same vendor.

I need some samples of cards in order to build a better detection for them.

micolous, 2016-12-28:
https://github.com/vpereira/bilhete seems to have a few dumps.

There's a lot more data on the card which the reader doesn't decode, but not enough notes to be useful :(  It only reads the balance.

There doesn't appear to be _any_ standard header on these cards. They also use standard checksumming on records, so anyone who does the same thing will also be mis-detected.  One of the card dumps listed has different manufacturer data, so even some BÚ cards will be not detected as BÚ.

This will require that certain cards be set as a "fallback" reader in order to work properly.  Looking at the other MFC readers, they implement more robust checks so aren't affected by this problem.

micolous, 2016-12-28:
The BÚ reader is now disabled by default, and requires explicit turning on to work.

However, there seems to be a lot more data in the card than what is handled now, so this is turning into a feature request.

micolous, 2017-01-13:
Annotated dumps are required to fix this -- bear in mind I don't speak Portuguese so a lot of existing resources aren't helpful for me. :(

phcoder, 2018-09-27:
This can be closed now

closed 2018-09-30 by micolous