metrodroid/metrodroid #28: App Crashing

steets250, 2016-12-29:
- [x] I'm using the current release version of Metrodroid, or current `master`.

Nexus 7 (ME370T) - Version 5.1.1 when reading Transit Access Pass.

The app is crashing when I try to view the history, subscription, or info pages once I have scanned the TAP card. I attached the log file from after loading the card to when the crash occurs.
[log.txt](https://github.com/micolous/metrodroid/files/676466/log.txt)

micolous, 2016-12-29:
Can you please send me an email with the card dump?

I think I know the cause of this, I just need to double check that it's _actually_ fixed.

micolous, 2016-12-30:
Confirmed with the card dump that these patches resolve the issue.

closed 2016-12-30 by micolous