metrodroid/metrodroid #30: ITSO support (UK-wide transport smartcards)

kw217, 2017-07-15:
(Disclaimer: this is not a request for support, but just a note in case anyone else wants to work on this. I have read the New cards wiki page. Sadly I don't have time to work on this myself, but I wanted to record my research in case it was useful to others.)

If anyone is interested in working on support for cards in the United Kingdom, many/most operators (including ScotRail, my local operator) use the ITSO specification. This specification is [freely available](https://www.itso.org.uk/services/specification-resources/the-itso-specification/itso-technical-specification/) and quite detailed.

At one time @rjmunro did some work on ITSO support in farebot: https://github.com/codebutler/farebot/pull/56 which may serve as a basis for this work.

I'm happy to assist with dumps etc from my ScotRail smartcard if that's of use.

There is a closed-source app on Google Play (Rambus Smart Ticket Checker) which supports these cards, but it would be good to have this code in the open.

micolous, 2017-08-07:
I'm happy if anyone wants to pick this up, but as I don't live in the UK it is not possible for me to do this.

As a side note, as I understand the ITSO specification is _not_ compatible with London's Oyster card.  The DESFire-based Oyster card I've seen has no freely readable data.  The older Mifare Classic-based cards might be readable with the Cubic Nextfare reader (as it [might be the same system](https://www.cubic.com/Portals/0/Publications/IntroducingCubic-NorthAmerica.pdf)).

micolous, 2017-10-29:
[According to the spec](https://www.itso.org.uk/download/42069.3/part-10-customer-media-definitions/), there appears to be ways to encode ITSO data on multiple card media, including Mifare Classic, Ultralight, DESFire, Calypso, and a generic EMV interface that appears to be similar to PayPass/PayWave.  This is going to be quite hard to implement because of the different media involved.

I'm also not sure if there's enough information in the public domain to actually implement this without detailed visits -- there's a section of the docs which is only available under NDA, which is something that isn't really "compatible" with an open source application (that also has implementation notes available in the public domain).

This may be something that's better off as some reference material rather than an issue that I'll never be able to properly resolve.

kw217, 2017-10-30:
Fair enough. I'm happy to close this issue - people can always find it by searching.

micolous, 2017-11-04:
Copied information here: https://github.com/micolous/metrodroid/wiki/ITSO

At the moment, I can't really resolve this unless someone else wants to pick this up again.

closed 2017-11-04 by micolous