metrodroid/metrodroid #31: OV-Chipkaart not working.

emveeha, 2017-08-08:
The card is a SLE66 emulating mifare classic.

I have succesfully cracked the A and B keys using:
https://github.com/aczid/crypto1_bs

I am able to make a dump with MFOC now. If I load this key into metrodroid and scan my card it still tells me I need the encryption keys. If I just make a plaintext file cointating the A and/or B keys I get the same result.

Let me know if you need the MFOC dump.

micolous, 2017-08-08:
I'm able to load your dump into Metrodroid manually fine -- the balance on the card is €12.60 and was last used on 2016-01-12.

Loading it manually:

1. `python extra/mfcdump_to_farebotxml.py -o ovchip.xml mfoc.mfd`
2. Copy `ovchip.xml` to the device
3. Open Metrodroid and click "History"
4. From the menu, select `Import` -> `File`
5. Select `ovchip.xml` in file browser

It seems like you've loaded the keys in the incorrect format.  The keys must be encoded in raw (binary) form, not text form.

Getting your keys in the right format:

1. `python extra/mfcdump_keys.py -o ovchip.farebotkeys mfoc.mfd`
2. Copy `ovchip.farebotkeys` to the device
3. Open `ovchip.farebotkeys` in file browser
4. Pick `Add Keys`
5. Tap your OV-Chipkaart to your device
6. Select `Key A`
7. Select `Add Key`

You'll then be able to read the card with your phone.

emveeha, 2017-08-08:
Loading manually worked (in gplay version)! Pretty fun.

Sadly reading the OV-Chipcard with the device is not working still. Output:
https://pastebin.com/80QuBG9m

BTW: loading manually did not work on the fdroid version:
error: AttributeException: Attribute key does not have a match in class .....ClassicSector at line 3

micolous, 2017-08-08:
The FDroid version is out of date. The error you're seeing is because it's falling through to the MyWay/SmartRider reader which detects based on keys.  There are supposed to be automatic updates, but that's broken and I think it's to do with an issue [we discovered with FDroid when I was initially submitting the build](https://gitlab.com/fdroid/fdroiddata/merge_requests/1921#note_20759903). 

As for your read failure, it looks like there's an authentication failure on the card for sectors 0-21 inclusive.  While most of the data is stored in the other sectors, the [Magic which identifies OVC is stored in sector 0](https://github.com/micolous/metrodroid/blob/master/src/main/java/com/codebutler/farebot/transit/ovc/OVChipTransitData.java#L211).

I've noticed some Mifare Classic cards refuse authentication even when the key is correct, so try increasing "Authentication Retries" in Metrodroid settings.  When you tune up authentication retries, it will take significantly longer to give up, and will try every other sector key and default key on each "retry" (so each retry = ~43 authentication attempts for a MFC 4K card).

Looking at your dump, a key of all `00` is used for that sector, which is included by default with Metrodroid and is also used in a lot of other sectors in your keyfile.  Android always tries to authenticate with the MAD key before any application gets to talk to the card... which doesn't help the situation a whole lot when the card isn't behaving.

emveeha, 2017-08-08:
Thank you for your fast and detailed response.

After increasing the "Authentication Retries" in Metrodroid settings to 20 it still fails to read the card.
This is all done with a OnePlus 5. A Moto X 14 does not read the card at all.

Hard coding the OVC or Magic could be a (nasty) solution right?

micolous, 2017-08-10:
Sorry, I don't have a good solution for you. 😞 

The retry system normally helps, but it sounds like there's a problem with the NFC stack on the OnePlus 5 which is preventing it from authenticating correctly when there is a key of all `00 00 00 00 00 00` on a Mifare Classic card.

On your card dump, the key set on sectors 0 - 21 is `00 00 00 00 00 00`.  Sector 0 has some other content, and sectors 1 - 21 have only bits for Mifare Classic access control in them (`07 87 8F 00`).  The block layout is described in more detail [on this document, starting at Page 6](http://www.nxp.com/docs/en/application-note/AN1304.pdf).

This could be confirmed if:

1. You tried a different OVC card and observed the same behaviour, or,
2. You changed the key on sectors 0 - 21 to something else and it worked, or,
3. You tried to read another Mifare Classic 4K card which had keys set to `00 00 00 00 00 00` (Metrodroid would not recognise it and complain, but it would at least report an authentication failure / fully locked card).

Note that changing the key would _prevent you from using the card_ with a legitimate reader until you reverted it to the correct code.

I have neither an OVC card nor a OnePlus phone, so I can't do much troubleshooting here.  Actually fixing the issue would require the intervention of the manufacturer. 😞 

**In the case of the OVC reader**, it doesn't look like sector 0 is used for anything except identifying the card.  So if I patched the Mifare Classic Fallback functionality to allow using the OVC reader, it may be an option to work around issues on your phone with this particular card.

**Regarding your Moto X 2014**, I have no idea what NFC chipset it has.  Consult the about screen and look for "Mifare Classic Supported".  The Moto X 2013 reportedly had a NXP chip which should support it, but that's no guarantee that Motorola used NXP again.

Mifare Classic cards aren't "standard" NFC Forum tags, so can only be read with devices that have NXP chipsets.

emveeha, 2017-08-10:
I have think I have confirmed the problem regarding OnePlus NFC stack:

1. A non OVC with an default key off `FF FF FF FF FF FF` throws an orange error:
`This card is not yet supported by Metrodroid. Only raw data can be displayed`
And most sectors are readable.

2. A fully locked OVC with key set to `00 00 00 00 00 00` throws an red error just like the unlocked one:
`Encryption keys are required to read this card.`
As excpected all sector are (Unauthorized)

Regarding changing key on sector 0-21. How can I do this? I guess I need to edit my mfoc dump and write it back, how can I edit the dump?

Can a custom rom help? Thanks!

micolous, 2017-08-14:
Not quite ready yet, that was just the first part of what I need to patch...

micolous, 2017-08-14:
Hmm, actually, looked more into patching this... it looks like it isn't possible to do what I thought.

The OVC reader calls `mPreamble = parser.getPreamble();`.  This in turn calls `byte[] data = mCard.getSector(0).readBlocks(0, 3);` (reading sector 0) in order to populate a bunch of data in `OVChipPreamble`, including the expiry date and card type.

As for writing stuff back to the card, you can [use the `libnfc` tools to write back a modified dump](http://manpages.ubuntu.com/manpages/zesty/man1/nfc-mfclassic.1.html).  However, I haven't attempted that before.

emveeha, 2017-08-15:
Bummer. Let me know if there is anything I can do to help.
Your help is much appriciated.

micolous, 2017-09-03:
Sorry, I can't fix this. It is something that will need to be fixed by OnePlus.

I recommend reaching out to their support channels for more help with this, they'll need to write a patch to the NFC libraries on the phone and push out an update.

closed 2017-09-03 by micolous

emveeha, 2017-09-05:
That is unfortunate. Thanks for trying though.

ikarus23, 2017-11-17:
I just stumbled across this issue looking for something else. Do I get this right? The OnePlus 5 is not able to read/write Mifare Classic tags if they have all zero keys?

I'm developing the [Mifare Classic Tool](https://github.com/ikarus23/MifareClassicTool/) and some users are experiencing wired behaviors when they try to read tags with all zero keys. However they did not use a OnePlus 5 but a Sony Xperia Z3 Compact or a Sony Xperia XA Ultra. The issues can be found at https://github.com/ikarus23/MifareClassicTool/issues/66 and https://github.com/ikarus23/MifareClassicTool/issues/129

Could it be that the behavior of the Sony devices is connected to the OnePlus 5?

micolous, 2018-02-08:
Yup, looks like it.