metrodroid/metrodroid #36: Opal employee card not readable

philcolbourn, 2018-01-26:
Please read the guidelines for contributions (CONTRIBUTING.md) before submitting new issues to this tracker. Failure to adhere to these guidelines may cause offense.

## Unsupported card requests

If the card is **not** presently supported by Metrodroid:

- [ ] [I have read and acknowledged the New cards wiki page](https://github.com/micolous/metrodroid/wiki/New-cards)
- [ ] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [ ] I have attempted to understand the card format on my own.
- [ ] I have provided notes on what I've found so far.

## Description of the issue

I have an Opal good card which looks same as Opal employee card.

I get IOexception: tranceive failed

## For card-specific issues

- Name of the card type: opal
- Name of the transit operator: transport for nsw
- Location: ?

For all issues reading a _currently supported card format_ (eg: balance shown is incorrect), please export a card dump from Metrodroid and email it to me privately.

## Steps to reproduce

1. Tap card to phone
2. 
3. 

## Expected behaviour

## Actual behaviour
Error as above

## Workarounds (if any)

## Logs

Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

Please do not attach it to this issue. You may be asked for a copy later via private email.

## Details from About screen:

- Version:2.9.3(32)
- Model:nexus 6p (angler)
- Manufacturer:huawei
- Android OS version:8.1.0
- NFC: enabled
- Mifare Classic: supported

micolous, 2018-01-26:
This is partially fixed in git master (there's an issue with reading locked
Mifare Ultralight cards that are locked), but Mifare Ultralight C cards
aren't being detected correctly.

I'm playing with the paper Single Trip tickets to get them detected
properly (as Ultralight C), but I suspect that Employee cards will sit into
the same category as SSTS cards (and not readable until I have a cracked
card).

micolous, 2018-01-28:
I think I've addressed the issue in b621f9e.  There's a now a lot of new code in there to handle Mifare Ultralight cards better, as it looks like Android's implementation is lacking.  This will be in the next version of Metrodroid (2.9.33).

The single use cards (and I presume, free travel passes will be the same) appear to be fully locked.  When scanning one, Metrodroid 2.9.33 will now return a "fully locked card" error instead of IOException.

The public Opal Travel app doesn't appear to support these either.

As there doesn't seem to be a whole lot released in the Mifare Ultralight C cracking department at present, it isn't possible for me to get at some useful data for these tickets yet.  This situation may change in the future.

I haven't implemented Mifare Ultralight C/EV1 authentication, and Android doesn't appear to natively support it either.

micolous, 2018-03-17:
Closing this as v2.9.33 is out, which no longer errors this way on Opal Ultralight C cards.  You now get "fully locked card".

closed 2018-03-17 by micolous