metrodroid/metrodroid #40: Pixel 2 (and XL) reports "no MIFARE Classic support" when it supports MIFARE Classic.

micolous, 2018-02-08:
## Description of the issue

Pixel 2 (walleye) and Pixel 2 XL (taimen) report "no MIFARE Classic Support" in Metrodroid, when they actually support it.

iFixit reports "NXP 81A04 39 04 sSD730 (likely NFC controller) and Murata SS7715005 (Wifi/Bluetooth module?)" for Pixel 2 XL.

Pixel 2 is also able to communicate with MFC cards just fine.

This is probably due to the extra detection routines added for the LG F60 (issue #26) picking up something it doesn't like.

micolous, 2018-02-08:
Analysis process (useful for debugging other phones)

1. `apt install simg2img abootimg liblz4-tool`
2. [Downloaded Pixel 2 and 2 XL factory images](https://developers.google.com/android/images#taimen).
3. Unzip file
4. `mkdir img; cd img`
5. `unzip ../image-*.zip`
6. `simg2img system.img system.ext4`
7. `simg2img vendor.img vendor.ext2`
8. `mount -o ro system.ext4 /media/android`
9. `mount -o ro vendor.ext2 /media/android/vendor`

Then start poking at the filesystem... `sudo find /media/android2/ | grep -i nfc`

Getting the kernel:

```
abootimg -x boot.img
unlz4 -d zImage kernel
```

micolous, 2018-02-08:
Looks like the NFC controller inside the Pixel 2 and Pixel 2 XL is an NXP PN80T or PN81A. There are, however, traces of the Broadcom 2079x driver in the filesystem image, specifically `/vendor/etc/libnfc-brcm.conf`.  This is also true of the Pixel 1.

And, the plot thickens, the LG F60 supports NFC only with 1 variant (e2n).  That also uses a PN544 or PN547 chipsets: https://github.com/TeamRegular/android_device_lge_e2nxx-common/blob/cm-12.1/rootdir/etc/ueventd.qcom.rc   But I can't find a factory system image for that phone.  But it's quite old, so I'm not sure how much I'm worried about it.

The check was inherited from Mifare Classic Tool.  It looks like it breaks in other ways: https://github.com/ikarus23/MifareClassicTool/issues/152

micolous, 2018-02-08:
I've disabled this check in 5c487da2034577e5e7300cd446fb5abfa374a7cf.

closed 2018-02-08 by micolous

GabrielTK, 2018-03-16:
Sorry, I don't know if I forgot to do something, but I'm having the same problem.

DavidBerdik, 2019-02-04:
MIFARE Classic is also listed as not supported on here: https://www.shopnfc.com/en/content/7-nfc-compatibility

Does anyone know about the status for the Pixel 3 XL?

micolous, 2019-06-13:
ShopNFC marks several MIFARE Classic-supporting phones as "no".

It doesn't even line up entirely line with `com.nxp.mifare`, so I suspect they're using that as a first source, and then respond to their customers when they report unexpected success.  However, if you were told that a NFC tag is not going to work with your device(s), you probably wouldn't buy it on the off-chance, so it is erring on false negatives.

I've looked at the [Pixel 3](https://android.googlesource.com/device/google/crosshatch/+/android-9.0.0_r21/nfc/libnfc-nxp.blueline.conf#271) and [Pixel 3 XL](https://android.googlesource.com/device/google/crosshatch/+/android-9.0.0_r21/nfc/libnfc-nxp.crosshatch.conf#270), and like the Pixel 3A (and 3A XL), it seems that it also has the exact same bug.  1b0f7d33 will work around this.

If **Metrodroid** incorrectly reports another phone as not supporting (or supporting) MIFARE Classic (and you've verified it actually works), please open a new bug, and I'll add it to the list.

The issue is largely cosmetic -- you can always install Metrodroid, and it doesn't block any code paths if MIFARE Classic support was not detected.

Ultimately, _these are all bugs from your device vendor and/or Android distribution vendor._  This [is part of the Android CDD](https://source.android.com/compatibility/9/android-9-cdd.html#7_4_4_near-field_communications) (`C-4-1` and `C-4-2`).