metrodroid/metrodroid #42: Propose New Logo

anaufalm, 2018-07-06:
Hello, I want to contribute to this awesome project, and I have a new logo design for you, if you allowed I will send to you :)

micolous, 2018-07-07:
Sure, feel free to attach/PR, so I can take a look.  I'm happy to put it in the right places for you.

If you've got vectors for it as well, that's better, because then it generally saves space and can be scaled up/down to support different resolution displays.

The one that's there now is based on Sydney's complete public transport network, using the actual geometry, rather than a [Harry Beck-style design](https://en.wikipedia.org/wiki/Harry_Beck) that is common with most networks today.

anaufalm, 2018-07-07:
Hi, here is my proposal, what do you think? please give me your opinion.
Some image inspiration: [train](http://www.traintosuratthani.info/wp-content/uploads/2016/11/15.35-Train-to-Surat-Thani.jpg), [metro map](http://www.johomaps.com/oc/australia/nsw/sydney/g_sydneymetro1.jpg)

[image]

micolous, 2018-07-14:
Looks good!  Just so you know, that "metro map of Sydney" is fictional. ;)

Let me have a play with this and get back to you.

anaufalm, 2018-07-15:
Ohh I see, but is there something happen with the logo? Please give me your opinion, so we can make it better :)

micolous, 2018-07-15:
I like the dark green "material" look one (3rd from bottom).

There is nothing wrong with it... I just wanted to experiment with the vector, in order to come up with something to replace [the "title" card image](https://github.com/micolous/metrodroid/blob/master/src/main/res/drawable-xhdpi/logo.png) and the variation used for [the Play Store feature graphic](https://developer.android.com/distribute/best-practices/launch/feature-graphic).

anaufalm, 2018-07-15:
This is the [file](https://drive.google.com/open?id=1Zmu8HTv8sju21JC2WqUqs2MHKDlPWfTl), if you wanna use it, also vector file in .pdf. Happy experiment :)
Dont forget to tell me if there is something about the logo.

micolous, 2018-08-18:
I tweaked it a little to fit in Material's [keyline shapes](https://material.io/design/iconography/product-icons.html#grid-keyline-shapes), snapping to the grid, and made most of the corners non-rounded.

I'm still tweaking to figure out the rest of the places it can be used.  My current thought is to get rid of the big logo on the main screen and replace it with something smaller in the corner.  But this is proving to be tedious on the code side of things...

micolous, 2018-08-19:
Icon in menu list:

[image]

New home screen so far:

[image]

What I'm planning to do with the home screen:

[image]

anaufalm, 2018-08-19:
Hi, I've update the [folder](https://drive.google.com/open?id=1V3M_R02xP94QOVdt1Y51pi0v5TUr_ckw) with non-rounded corner logo. Please check it.

And about the home screen design, can you explain more, what we want to create? I see the detail that you wrote is cropped, so I cant read it.

naufaldsg, 2018-12-16:
May I upload new icon to /src/main/res/mipmap @micolous? I will create pr

micolous, 2019-05-19:
Heya, sorry, this fell off my radar.

I've got to rebuild the branch where I was working on this anyway, because it is now far behind what `git master` has.  I can update that when I get there.

micolous, 2019-07-23:
Thanks for this, but for now, I want to keep the current logo for the near future, and close this out.

I want to focus on other pressing issues right now, and then come at logo / style / UI changes with a clear head and a planned approach.

closed 2019-07-23 by micolous