metrodroid/metrodroid #149: clipper: investigate agency IDs for Walgreens

micolous, 2018-08-27:
## Description of the issue

The [Clipper operators list](https://github.com/micolous/metrodroid/blob/master/data/clipper/operators.csv) has some interesting bits.  It looks like Walgreens has 0x1bc, 0x1be, 0x1db.

Given the spread, there may be many more operator codes here, and they may represent an individual Walgreens location.

## For card-specific issues

- Name of the card type: Clipper
- Name of the transit operator: SFMTA
- Location: San Francisco, CA, US

mcpherrinm, 2018-09-22:
I scanned a bunch in the past, and it seemed like each Walgreens location I tested had a seperate ID

micolous, 2018-09-23:
Thanks, good to know! Sounds like there is the possibility to add Walgreens locations to the database.

If folks know about a Walgreens (or other retailer) ID that is missing, please add a comment here. If you know the address of the store, please also note it here.

I wonder if SFMTA customer service centres, SFMTA stations, Whole Foods and other retailers are also set up this way. Because there's "SFO International Airport Booth".  Almost makes me think every retailer has a unique ID.

Last time I was in SF, I was to buy a new Clipper and top-up with 1.00$ at Walgreens.  The clerk didn't mention any minimum recharge amounts, but I'm pretty sure humans would get annoyed if I wanted to top up my card with 0.01$.

stefanor, 2020-02-16:
FWIW: https://docs.clippercard.com/retail/locations_PROD.xml (but no clipper IDs in there, of course)