metrodroid/metrodroid #166: GP: Adding a key can cause OOM

micolous, 2018-09-09:
## Description of the issue

```
java.lang.OutOfMemoryError: 
  at org.apache.commons.io.output.ByteArrayOutputStream.needNewBuffer (ByteArrayOutputStream.java:127)
  at org.apache.commons.io.output.ByteArrayOutputStream.write (ByteArrayOutputStream.java:158)
  at org.apache.commons.io.IOUtils.copyLarge (IOUtils.java:2147)
  at org.apache.commons.io.IOUtils.copy (IOUtils.java:2102)
  at org.apache.commons.io.IOUtils.copyLarge (IOUtils.java:2123)
  at org.apache.commons.io.IOUtils.copy (IOUtils.java:2078)
  at org.apache.commons.io.IOUtils.toByteArray (IOUtils.java:721)
  at au.id.micolous.metrodroid.activity.AddKeyActivity.readKeyFile (AddKeyActivity.java:168)
  at au.id.micolous.metrodroid.activity.AddKeyActivity.onCreate (AddKeyActivity.java:140)
  at android.app.Activity.performCreate (Activity.java:7183)
  at android.app.Instrumentation.callActivityOnCreate (Instrumentation.java:1220)
  at android.app.ActivityThread.performLaunchActivity (ActivityThread.java:2908)
  at android.app.ActivityThread.handleLaunchActivity (ActivityThread.java:3030)
  at android.app.ActivityThread.-wrap11 (Unknown Source)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1696)
  at android.os.Handler.dispatchMessage (Handler.java:105)
  at android.os.Looper.loop (Looper.java:164)
  at android.app.ActivityThread.main (ActivityThread.java:6938)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:327)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1374)
```

Only reported on 2.9.34.

micolous, 2019-05-04:
This probably still exists, but this is caused by loading a too large file.  We should be more defensive.

AmoledK, 2023-05-22:
Can you provide a sample Mifare classic Key file format ?? i have tried with json. but i am getting unknown fileformat exception