metrodroid/metrodroid #167: Cannot read CEPAS / EZ-Link

sybond, 2018-09-11:
## Description of the issue
Please see attached [log](https://github.com/micolous/metrodroid/files/2368972/metrodroid_cepas.log).

phcoder, 2018-09-11:
Was it working before?  I'll prepare exploratory version. I'll read the standard again but I think it's silent on the issue of selection. (Yes, I bought the standard)

micolous, 2018-09-11:
I'll have to leave this one with @phcoder.

If you want to see if #159 is a probable cause, you want to roll back to https://github.com/micolous/metrodroid/commit/efc06922066b70d50f9823ea008fdb7749810e3c

sybond, 2018-09-12:
@phcoder working with latest release version.

phcoder, 2018-09-17:
Please try the branch cepasexp: https://github.com/phcoder/metrodroid/tree/cepasexp

sybond, 2018-09-17:
Here is the log:
```
09-17 09:43:40.722 18913-19043/au.id.micolous.farebot D/au.id.micolous.metrodroid.card.Card: Reading tag f2bb7264. 2 tech(s) supported:
    android.nfc.tech.IsoDep
    android.nfc.tech.NfcB
09-17 09:43:40.754 18913-19043/au.id.micolous.farebot W/DesfireCard: Card responded with invalid response, may not be DESFire?
    java.lang.IllegalArgumentException: Invalid response: 6d00
        at au.id.micolous.metrodroid.card.desfire.DesfireProtocol.sendRequest(DesfireProtocol.java:163)
        at au.id.micolous.metrodroid.card.desfire.DesfireProtocol.getManufacturingData(DesfireProtocol.java:73)
        at au.id.micolous.metrodroid.card.desfire.DesfireCard.dumpTag(DesfireCard.java:106)
        at au.id.micolous.metrodroid.card.Card.dumpTag(Card.java:98)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:191)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:182)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
09-17 09:43:40.790 18913-19043/au.id.micolous.farebot D/au.id.micolous.metrodroid.card.iso7816.ISO7816Protocol: Select file 4000
09-17 09:43:40.818 18913-19043/au.id.micolous.farebot W/CEPASProtocol: Error reading purse 0
    au.id.micolous.metrodroid.card.iso7816.ISO7816Exception: Got unknown result: 6b00
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Protocol.sendRequest(ISO7816Protocol.java:131)
        at au.id.micolous.metrodroid.card.cepas.CEPASProtocol.getPurse(CEPASProtocol.java:43)
        at au.id.micolous.metrodroid.card.cepas.CEPASApplication.dumpTag(CEPASApplication.java:109)
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Card.dumpTag(ISO7816Card.java:106)
        at au.id.micolous.metrodroid.card.Card.dumpTag(Card.java:103)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:191)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:182)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
09-17 09:43:40.824 18913-18919/au.id.micolous.farebot I/zygote64: Compiler allocated 8MB to compile void android.widget.TextView.<init>(android.content.Context, android.util.AttributeSet, int, int)
09-17 09:43:40.838 18913-19043/au.id.micolous.farebot W/CEPASProtocol: Error reading purse 1
    au.id.micolous.metrodroid.card.iso7816.ISO7816Exception: Got unknown result: 6b00
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Protocol.sendRequest(ISO7816Protocol.java:131)
        at au.id.micolous.metrodroid.card.cepas.CEPASProtocol.getPurse(CEPASProtocol.java:43)
        at au.id.micolous.metrodroid.card.cepas.CEPASApplication.dumpTag(CEPASApplication.java:109)
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Card.dumpTag(ISO7816Card.java:106)
        at au.id.micolous.metrodroid.card.Card.dumpTag(Card.java:103)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:191)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:182)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
09-17 09:43:40.855 18913-19043/au.id.micolous.farebot W/CEPASProtocol: Error reading purse 2
    au.id.micolous.metrodroid.card.iso7816.ISO7816Exception: Got unknown result: 6b00
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Protocol.sendRequest(ISO7816Protocol.java:131)
        at au.id.micolous.metrodroid.card.cepas.CEPASProtocol.getPurse(CEPASProtocol.java:43)
        at au.id.micolous.metrodroid.card.cepas.CEPASApplication.dumpTag(CEPASApplication.java:109)
        at au.id.micolous.metrodroid.card.iso7816.ISO7816Card.dumpTag(ISO7816Card.java:106)
        at au.id.micolous.metrodroid.card.Card.dumpTag(Card.java:103)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:191)
        at au.id.micolous.metrodroid.activity.ReadingTagActivity$ReadingTagTask.doInBackground(ReadingTagActivity.java:182)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[... cut ...]
09-17 09:43:41.194 18913-18913/au.id.micolous.farebot I/Timeline: Timeline: Activity_launch_request time:28819623
09-17 09:43:41.356 18913-18913/au.id.micolous.farebot I/Timeline: Timeline: Activity_launch_request time:28819785
09-17 09:43:43.396 18913-18919/au.id.micolous.farebot I/zygote64: Do partial code cache collection, code=107KB, data=106KB
    After code cache collection, code=107KB, data=106KB
    Increasing code cache capacity to 512KB
09-17 09:43:43.397 18913-18919/au.id.micolous.farebot I/zygote64: Compiler allocated 6MB to compile void android.view.ViewRootImpl.performTraversals()

```

phcoder, 2018-09-17:
I updated cepasexp with another fix. Turns out that CEPAS is subtly different in zero-parameter commands (or our generic wrapper is wrong)