metrodroid/metrodroid #262: Calypso 1.0 - ISO14443 Type B card not recognized

ernyugh, 2018-12-06:
## Description of the issue
I have a ISO14443B card that uses the Calypso 1.0 protocol. I'm sure of this because I've read official documentation from the company that states witch card they use.
If I put the card on my ACR122U and give nfc-list command, it says there is such ISO14443B card, but mfoc does not found any tag.

Now the issue:
If I scan the tag with the debugger attached to android studio, it does not even pass from onNewIntent(Intent intent) in ReadingTagActivity, so it's like I haven't approach anything.
But... Metrodroid actually supports Calypso! Could this be related to phone specific NFC stack?
I have an Oneplus 6, but I tried also with Oneplus 3T and LG G7.
Someone could read any ISO14443B Calypso card with metrodroid?

## Steps to reproduce

1. Approach a ISO14443B Calypso 1.0 card with metrodroid

## Expected behaviour
onNewIntent(Intent intent) being called

## Actual behaviour
onNewIntent(Intent intent) is not called

## Possible solution
I saw in res/xml/filter_nfc.xml that is missing NfcB and in the intent it does not expect NfcB, but even in this way, I can't recognize the card. It's like it is not supported... by android?

## Details from About screen:

- Version: 2.9.36-269-ge4204e37-dirty (1740)
- Model: ONEPLUS A6003
- Manufacturer: OnePlus
- Android OS version: 9.0
- NFC: enabled
- MIFARE Classic: supported

phcoder, 2018-12-06:
Some Calypso cards and all SRI cards are not readable by phones I have. E.g. some Navigo cards are readable and some are not. It looks like some kind of problem with ISO14443B select sequence. Unfortunately since it happens before Metrodroid is involved I don't think there is much we can do except possibly try reader mode. It seems to be an issue with NFC chipset, NFC stack or Calypso compliance with iso14443 B. Lack of NfcB in intent filter is a nice find but IsoDep should be enough.

ernyugh, 2018-12-07:
Ok, so it's not a metrodroid fault but a NFC stack one.
I can try to enable reader mode then manually send the SELECT sequence to the tag.
I have used this code:
`options.putInt(NfcAdapter.EXTRA_READER_PRESENCE_CHECK_DELAY, 5000);
adapter.enableReaderMode(this,
                new NfcAdapter.ReaderCallback() {
                    @Override
                    public void onTagDiscovered(final Tag tag) {
                        recievedInput();
                    }
                },
                NfcAdapter.FLAG_READER_NFC_A | NfcAdapter.FLAG_READER_SKIP_NDEF_CHECK | NfcAdapter.FLAG_READER_NFC_BARCODE | NfcAdapter.FLAG_READER_NFC_B | NfcAdapter.FLAG_READER_NFC_F | NfcAdapter.FLAG_READER_NFC_V | NfcAdapter.FLAG_READER_NO_PLATFORM_SOUNDS,
                options);`
where I am using all the NFC flag but nothing happens, I am able to find all tags except the ones I am working to.

Were you referring to this while stating 
> we can do except possibly try reader mode
or you meant something different and deeper in the NFC stack?

phcoder, 2018-12-13:
I meant trying enabling rewader mode and playing around. Similar to what you tried but also issuing custom SELECT commands. I didn't try myself but found following interesting pieces:
"In the same time, the international standard ISO/IEC 14443 for contactless smart cards was being designed, and the actors of Calypso strongly lobbied to have their technology included in the standard, but Innovatron's patents—and the price of the related royalties—were not compliant with ISO's policy. Therefore, despite their closeness, there are few significant differences between Calypso's historical contactless protocol and ISO/IEC 14443 Type B international standard." [from Wikipedia]
"Revision 2: Full ISO/IEC 14443 type B protocol" [CALYPSO FUNCTIONAL SPECIFICATION].
So it looks like Calypso 1.0 is not compatible with most phones while Calypso 2.0 is.
Calypso 1.0 can be read through contact interface or through devices that allow deeper protocol access, like proxmark3. For contact interface I recommend cheap smartcard readers e.g. USB readers from China like Rocketek one and cardpeek for dumping, it already has Calypso script. It's feasible to add USB-CCID support to Metrodroid, shouldn't be too hard.
So far I was unable to find a phone reading Calypso 1.0. It's possible that by putting phone in reader mode and issuing custom SELECT command it will work.
It also seems that there is some corellation between card and Calypso version but it's not strict.
So far I've seen:
- 2x Rav-Kav of newer revision (2018 card)
- 1x Pastel of newer revision (unsure but probably ca 2016)
- 1x Envibus of old revision (ca 2011 card)
- 1xNavigo of old revision and 1x Navigo of new one
- 1xVenezia Unica of newer revision (2018 card)
- 1xPisa Carta Mobile: compatible with ISO14443B but with some protocol weirdness
- 3xOpus of newer revision
- 1xOura of old revision (2018 card)

There is also a similar problem with SRI cards. They're essentially Ultralight counterpart in Calypso ecosystem and so far I managed to read them only with proxmark3.

micolous, 2018-12-16:
FYI, [I've got a branch where I implemented "reader mode"](https://github.com/micolous/metrodroid/tree/reader-mode), because I was trying to add support for cards served by Android HCE.  It's somewhat usable.  That might be useful for testing an Innovatron/baguette card, but I was still trying to stablise it.

To my knowledge, I don't have any Calypso 1.0 cards to try either -- because the Calypso cards I have "just worked" with the devices I've tried here.

I'm definitely interested in supporting USB readers, but will probably need to do a second build configuration to make that properly visible on the Play Store (ie: your device supports NFC **or** USB host).  I have a couple of PN532 attached to TTL serial adapters (purchased long ago), and an ACR122. That would also help with devices that don't support MIFARE Classic.

My experience so far with writing [AndProx](https://github.com/andprox/andprox) has shown me that USB host support on Android is messy.