metrodroid/metrodroid #357: Project Build Failed / Spanish Translate Proposal

NokisDemox, 2019-01-04:
## Description of the issue
I click on make project and it says "CreateProcess error=2"

I have git updated, github account fully working in android studio and updated JDK and I can't find more info about this error.
This application looks very interesting and I would like how it works and which keys uses.

Also in this same issue I would like to told you about translating this app to Spanish, im from Spain and I think I can help you guys. This app is not very known here and seems to have a lot of potencial, if i were able to build the project i will help.

I made a MOD apk for Mifare Classic Tool app and now im adding some new features to it, it will be possible to add your keys to that app?

Link to the video of the MOD apk: https://www.youtube.com/watch?v=Or2XoKedLS4

Or you can just look for "Mifare Classic Tool Bruteforce Android" in youtube,

Thanks for making such a good app 😁

micolous, 2019-01-06:
## Regarding the build issue

It sounds like you're building on Windows (because of `CreateProcess`), but I don't know why you get that error, and I'd need a stack trace from Gradle to know for sure.

I'd _guess_ it's because `git` is not in your `PATH` (at least, when running Android Studio).  You'll need to edit [your environment variables](https://www.java.com/en/download/help/path.xml) so that `git.exe` is actually in your PATH (and not just when using a "git bash" / "git shell" function), then restart Windows.  You'll know it's working when `git` works from a plain old command prompt (Win+R -> `cmd` -> OK -> `git` -> Enter)

I've honestly not attempted to build Metrodroid on Windows, so it might be broken in other ways.  I've got it working on Linux and OSX.

## Translations

If you're interested in translating the app, that's all [handled on Weblate](https://hosted.weblate.org/engage/metrodroid/). That tool is entirely web-based, so you don't need a working Android build environment.

There's no support for any transit provider in Spain, so it might not be a lot of use to you.  If you have some more information, then that would be helpful.

## Keys

The project only has 4 keys in it, 3 of which are provided by the Android platform, which you can find here:

https://github.com/micolous/metrodroid/blob/51981dae35c1c1d8a2fb50224db92a7548b07506/src/main/java/au/id/micolous/metrodroid/card/classic/ClassicCard.java#L95-L118

https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/nfc/tech/MifareClassic.java#76

But, I'll disappoint you now, because _Metrodroid doesn't ship with other, transit-provider-specific keys_, and MIFARE Classic Tool already has those 4 keys.

Most transit agencies using MFC implement _key diversification_, so each card has unique keys derived from a one-way-hash of the card's UID and an agency-specific secret.  Those secrets aren't needed to get the key, as MFC has other security weaknesses, but this means a static key list isn't useful for those cards.

There are a small number of agencies who use all or some of the same keys on every card.  The transit providers with static keys that Metrodroid supports are in public lists elsewhere.  [MIFARE Classic Tool has quite a lot of keys, and while not all of them are labelled as such, but there are definitely some static transit card keys in there.](https://github.com/ikarus23/MifareClassicTool/tree/master/Mifare%20Classic%20Tool/app/src/main/assets/key-files)

I personally don't use Android NFC to crack keys, it's slow and very limited as to what you can actually do.  [Using Proxmark3 or even an NXP PN532 is more versatile](https://github.com/micolous/metrodroid/wiki/Cracking-keys) -- PM3 cracks "weak" MFC cards in under a minute, PN532 takes under an hour.  PM3 can also sniff communication with a reader that has the keys.

If you don't have one already, _Proxmark3 is definitely worth the money_ (~300 USD) if you're interested in playing with RFID.

phcoder, 2019-01-06:
* You can get a proxmark3 easy variant for about $60. It's generally not very different from proxmark3.
* Transit agencies: I've had in my hands so far 3 different cards from Spain:
** Madrid red card is fully-locked desfire for which no practical attack is known
** Alicante and Camp of Tarragona. Static keys MFC. I have some parsing code for them but so far I can only decode serial number. If you have those cards, contact me privately, I can send you keys. Dumps would be very helpful for me.

NokisDemox, 2019-01-07:
I will think about buying one of them. I haven't got any of the card u talk me about but everyone i have tested is fully locked, i tested them with my app which has lots of keys and nothing ;(

micolous, 2019-01-08:
**Re: Magnetic stripe adaptors,** there are some different designs floating around for microphone based adapters, but these are mostly for credit-card form factor. When some of the mobile-app point-of-sale companies got started up (eg: Square), they offered devices to do this (but had some crypto on the unit to try to stop others using the readers).  They've now been replaced by chip-compatible devices, so you might get lucky and be able to find some second hand.

The principles of all magnetic-stripe cards are all essentially the same, but I'm not familiar enough with the modulation or electronics there to be able to give you something more useful.

In all Australian cities, public transit agencies stopped using long-term magnetic stripe tickets 3+ years ago. Many of them don't offer magnetic-stripe tickets at all.  Because of the extra hardware requirements, and the usage of the ticket is normally physically printed on the ticket, I haven't tried to look at that.

**Re: MFC keys,** [there are _much_ better ways to access MIFARE Classic cards](https://github.com/micolous/metrodroid/wiki/Cracking-keys) than brute-forcing them, which even the PN532 can implement _some_ of the attacks.

Getting _older_ cards also helps, because some of the bugs got "fixed" partially, and it's impossible to patch old cards.  If you're lucky, your local agency or their system integrator will try to save a few cents and use some clone card (eg: FM11RF08), and unlike NXP chips, those don't have some of the "fixes".  If you're unlucky, you will have all hardened cards and need to extract keys from a reader by spoofing the card or sniffing a genuine transaction. 

Given you can't find it in any key lists, it's very likely that your local agency implements key diversification. In which case every single card you have will need to be individually cracked.

NokisDemox, 2019-01-08:
**Given you can't find it in any key lists, it's very likely that your local agency implements key diversification. In which case every single card you have will need to be individually cracked.**
Thats sound hard, but im only going to try with public transport which all of them should have the same key so maybe a PN532  or any product similar to proxmark will be great idea. I really appreciate all the helpfull tips and info you are sharing.

micolous, 2019-05-12:
Closing this off as the Spanish translation has started, and pretty sure that we can call the build issue an environmental problem.  If the Windows builds are still failing with `git` in `PATH`, please let me know.

closed 2019-05-12 by micolous