metrodroid/metrodroid #367: Scaning Suica card results in NullPointerException

brianclemens, 2019-04-11:
## Description of the issue
Scanning a Suica card results in this error:
NullPointerException: Attempt to invoke the virtual method 'byte[] net.kazzz.felica.lib.FeliCaLib$IDm.getBytes()' on a null object reference

## For card-specific issues
- Name of the card type: Suica
- Name of the transit operator: JR East
- Location: Japan

## Steps to reproduce
1. Scan Suica card
2. Enjoy error message

## Expected behaviour
1. Scan Suica card
2. Probably something that doesn't involve an error message

## Actual behaviour
1. Scan Suica card
2. Receive error message

## Logs
Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

Please do not attach it to this issue. You may be asked for a copy later via private email.

## Details from About screen:
- Version: v2.9.31-fdroid (31)
- Model: Nexus 6P (angler)
- Manufacturer: Huawei (google)
- Android OS version: 8.1.0 (OPM7.181205.001)
- NFC: enabled
- MIFARE Classic: supported

micolous, 2019-04-12:
Thanks for your report! Unfortunately, [the version of F-Droid is very out of date](https://micolous.github.io/metrodroid/fdroid).

What I think is happening for you is the same as #41: the card is dropping out of the reader's field, and then that results in a partial read of the card.  I made some changes, which are in 2.9.35 and later, which make it not give you back errors like this (and instead, report a "partial card read").

I see this with most of my FeliCa cards (the card media used by most Japanese transit cards) -- they seem to be particularly susceptible to dropping out.  I'm not sure whether this is caused by an issue with my code.  [I have added some more detail about the current status of FeliCa reading in #41](https://github.com/micolous/metrodroid/issues/41#issuecomment-482461126).

brianclemens, 2019-04-12:
New version works fine, sorry about that!

closed 2019-04-12 by brianclemens