metrodroid/metrodroid #373: New card request - Bilhete BOM

r-adamska, 2019-05-04:
Please read the guidelines for contributions (CONTRIBUTING.md) before submitting new issues to this tracker. Failure to adhere to these guidelines may cause offense.

## Unsupported card requests

If the card is **not** presently supported by Metrodroid:

- [ ] [I have read and acknowledged the New cards wiki page](https://github.com/micolous/metrodroid/wiki/New-cards)
- [ ] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [ ] I have attempted to understand the card format on my own.
- [ ] I have provided notes on what I've found so far.

## Description of the issue

There's a new type o mifare card on Brazil besides Bilhete Unico called Bilhete BOM. I have two of them working just fine, and I've broken the keys of both with mfoc. I have plenty of dumps, documented time, fare, date, and so on. I've already understood part of the logic, but there are some Value blocks that make some kind of checksum with Sector 2, that if we try to put a image of the day before on it, it gets a "integrity error" on the subway, so I have to put the normal back on, since I have no idea how to decrement Value Blocks on the card.

I'm using only Kali Linux for all of my research. I just need some help to understand how the integrity check works, and where is stored the fare value.

Looking forward to meet you!

## For card-specific issues

- Name of the card type: Bilhete BOM
- Name of the transit operator: EMTU
- Location: Sao Paulo - Brazil

For all issues reading a _currently supported card format_ (eg: balance shown is incorrect), please export a card dump from Metrodroid and email it to me privately.

## Steps to reproduce

1. 
2. 
3. 

## Expected behaviour

## Actual behaviour

## Workarounds (if any)

## Logs

Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

Please do not attach it to this issue. You may be asked for a copy later via private email.

## Details from About screen:

- Version:
- Model:
- Manufacturer:
- Android OS version:
- NFC:
- MIFARE Classic:

phcoder, 2019-05-04:
Please send dumps and your notes to us. If you want to do it privately just send them to @micolous and @phcoder (my nickname is my Gmail)

r-adamska, 2019-05-05:
Ok. I've sent you everything I got so far, please confirm that you've received.

micolous, 2019-05-07:
Yup, got it! I've got some follow-up questions about your dumps which I'll email you about.

What may also be of interest to you is this repository which has some notes about the older (BU) cards (written in Portuguese) and dumps (which Metrodroid's tests use): https://github.com/vpereira/bilhete/tree/master

Mrlluck, 2019-08-19:
Any progress on that?

micolous, 2019-08-24:
Closing this out as the emails were a request for help to defraud the operator.

closed 2019-08-24 by micolous