metrodroid/metrodroid #383: Clipper single-use tickets throw TagLostException in v2.9.38

iangcarroll, 2019-05-09:
Please read the guidelines for contributions (CONTRIBUTING.md) before submitting new issues to this tracker. Failure to adhere to these guidelines may cause offense.

## Description of the issue
Clipper single-use tickets fail to read in v2.9.38, throwing a TagLostException on my Pixel 3. I'm doing some research on these and have plenty of dumps if needed.

## For card-specific issues

- Name of the card type: Clipper (single-use)
- Name of the transit operator: Clipper
- Location: San Francisco, CA

For all issues reading a _currently supported card format_ (eg: balance shown is incorrect), please export a card dump from Metrodroid and email it to me privately.

## Steps to reproduce

1. Scan a Clipper single-use ticket (only tested with Muni-issued ones).
2. 
3. 

## Expected behaviour
Card reads, as it does in v2.9.37.

## Actual behaviour
TagLostException every time.

## Workarounds (if any)

## Logs

Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

Please do not attach it to this issue. You may be asked for a copy later via private email.

## Details from About screen:

- Version: 
- Model: Pixel 3 (blueline)
- Manufacturer: Google (google)
- Android OS version: 9 (PQ2A.190405.003)
- NFC: enabled
- MIFARE Classic: not supported

phcoder, 2019-05-10:
Try https://github.com/micolous/metrodroid/pull/384

closed 2019-05-10 by iangcarroll

micolous, 2019-05-12:
This will be fixed in v2.9.40.