metrodroid/metrodroid #398: Design Contribution

ghostofiht, 2019-05-13:
Hi, I am a designer. I support special projects with my design contributions. I see you have a good project. But unfortunately I wouldn't say the same for the logo. So I designed a logo for this project.
 
The main idea in design is;

 NFC bus tickets. And a bus with an NFC antenna inside the ticket. Because it is a simple, minimalistic logo, it will appear clearly in small android icon sizes. 

If there's something you want to change in design, I can do it. I'd like to inform you that this is a contribution. 

Here are the designs; 

[image]

micolous, 2019-05-19:
Heya, thanks for this.  _I agree the logo needs work_ -- this is something that I hacked together quickly in QGIS and Inkscape a while ago. :)

I have another outstanding contribution in #42 to change this logo, and had started integrating it with some other UI changes (noted in the bug).  I haven't been working on that lately, because I have been trying to merge some large structural changes in the code lately.

I like the bus coil concept, but my concern is that [the Play Store requirements have changed](https://developer.android.com/google-play/resources/icon-design-specifications), which now require uniform icon shapes (yes, exactly the opposite of what all the Google apps on your phone have...)

I'm not sure how to proceed on this at the moment, and need to think this over.

ghostofiht, 2019-05-20:
Hi again, I understand your concerns about the Logo format. According to the new rules, the logo form should be as follows.

[image]

512x512
32 bit PNG
sRGB

I can provide the final files according to these rules.  I can send a PR right away. However, you should choose a color from the designs.

ghostofiht, 2019-07-12:
Hey, @micolous , I think you've made a choice by this time. Please tell me your choice.

phcoder, 2019-07-12:
Does the logo complies with App store requirements as well? I'm writing now iOS port and it would be good to have the same icon

ghostofiht, 2019-07-13:
@phcoder Absolutely convenient. Compatible with current IOS application icon format. I will provide all the necessary files. All that remains is for @micolous  to choose a color.

micolous, 2019-07-13:
Heya, just a quick update – I haven't made a firm choice yet, and I'm waiting for a second opinion.

micolous, 2019-07-23:
Thanks for this, but I'm want to keep the current logo for the near future.

I want to focus on other pressing issues right now, and then come at logo / style / UI changes with a clear head and a planned approach.

closed 2019-07-23 by micolous