metrodroid/metrodroid #481: Questions on ATM (Milan metro card) implementation

Depau, 2019-06-16:
(tl;dr: skip to last part)

I have an ATM card (Azienda Trasporti Milanesi - Milan metro - not a bank card), I just saw [this](https://github.com/micolous/metrodroid/tree/master/src/commonMain/kotlin/au/id/micolous/metrodroid/transit/ricaricami) contributed by @phcoder, so I built the app from master and tested it.

First of all, the app also works with the regular ATM card (which looks like [this](https://www.atm.it/en/ViaggiaConNoi/Abbonamenti/PublishingImages/tessera.jpg)), not only with the RicaricaMi card (which is the one mentioned in the source code).

The main difference (to my knoweledge) is that the regular orange card is personal and can be used for monthly and annual subscriptions, while the ricaricami card is anonymous and can only be used to store short-term tickets. Most people living in Milan have the orange card, the green RicaricaMi card is a lot less common and is usually used by occasional travelers. That is to say I think "ATM card" in strings and source code would be better, though I see most people might confuse it with bank cards.

That being said I scanned my card (loaded with an annual subscription) and I noticed that
- Subscription name is displayed as unknown
  - As a matter of fact that makes sense and I see that in the source code, because the "RicaricaMi" card won't be sold with this kind of subscription
  - I think I might be able to patch it to support this
- Subscription issue date doesn't make sense to me (though it may be the issuer's fault, I'll check at a metro station totem next time I go there)
- I have some entries in my history and not all metro stations are known to the app
  - I haven't been able to find where in the code the station name is identified

### tl;dr here

Finally, my questions:
- Does it make sense to you to
  - Change strings and graphics and display it as "ATM Milano Card" or something like that?
  - Also refactor class/package names?
- Where the hell is the code that identifies the station names?

phcoder, 2019-06-16:
First of all thank you for you post.
I suspected that ATM card would also work but I needed a confirmation.
1) "Subscription name is displayed as unknown"
I see you already figured out how to add subscription names
2) "Subscription issue date doesn't make sense to me". RicaricaMi/ATM uses en1545 which can easily be misparsed when unknown fields are present. Can you take a picture of kiosk machine showing info on your car and the dump and send them both privately to me?
3) "Where the hell is the code that identifies the station names?" metrodroid/data/ricaricami/mapping.csv is probably what you'r looking for. After modification:
cd metrodroid/data/ricaricami
make
cp ricaricami.mdst ../../mdst/
4) "Change strings and graphics and display it as "ATM Milano Card" or something like that?" The same class can declare several supported cards. See "val allCards".
As for showing the card name ideally we would detect which card it is. Again I would need dumps or code for it
5) "Also refactor class/package names?" Meh, I don't care enough either way, I think it's fine to keep it as-is

Depau, 2019-06-17:
> 2. "Subscription issue date doesn't make sense to me". RicaricaMi/ATM uses en1545 which can easily be misparsed when unknown fields are present. Can you take a picture of kiosk machine showing info on your car and the dump and send them both privately to me?

I eventually noticed the date was reported in mm/dd/yyyy format so that was the problem. It's good ;)

> 3. metrodroid/data/ricaricami/mapping.csv is probably what you'r looking for...

Awesome. I'll send more patches soon.

> 4. The same class can declare several supported cards. See "val allCards".

I must say I had a real hard time navigating through the code, Android Studio refuses to recognize the `commonJvm` module and most features don't work.

That would be great as I also noticed you can tell Under26 card apart from regular ones, there's some profileId somewhere as the metro kiosk shows.

> 5. Also refactor class/package names?" Meh, I don't care enough either way, I think it's fine to keep it as-is

That works and it's probably less confusing.

micolous, 2019-06-17:
Thanks for the information! :)

> I eventually noticed the date was reported in mm/dd/yyyy format so that was the problem. It's good ;)

Yup, this uses the Android system locale.  Android _used to_ let you manually set this, but if you have the Locale set to `en_US` then you'll get middle-endian dates.

> 1. metrodroid/data/ricaricami/mapping.csv is probably what you're looking for...

Also easier for regenerating:

```
cd data
make ricaricami.build ricaricami.copy
```

Only caveat with working with this dataset is that downloading GTFS data from AMAT's website is extremely unreliable.  If you get stuck, [there's a copy in transitfeeds.com](https://transitfeeds.com/p/agenzia-mobilita-ambiente-territorio/341).

When working with that setup, it'll also dump `ricaricami.csv`.  This will let you see what's in the generated MdST file, and you can diff it with text-based tools.

Most of the `data` targets work in identical ways.

> I must say I had a real hard time navigating through the code, Android Studio refuses to recognize the `commonJvm` module and most features don't work.

It sounds like that things are blowing up a lot more than they should be.  

> > 1. Also refactor class/package names?
> 
> That works and it's probably less confusing.

I'm fine with that.  We sometimes refactor quite a bit when we learn new things. :)

---

My plan is to cut a release based on 2.9.41 sometime during this week, which will have the Ricaricami naming for now.  There's quite a bit of stuff in that release that I'd like to get out.

I'll get your first patch in the release after that (tentatively .42, if there is nothing very wrong in .41), and then depending on what time commitments you have and how we go with the next cycle, we may be able to get your other changes in too.

Depau, 2019-06-17:
Sounds great! I'll see if I can send the other patches today.

Depau, 2019-06-17:
Okay, I've seen the transit info scripts but I'm not sure how to add the mapping information for metro stations. I've seen that the ID is the station name in the downloaded data, however what are `reader_id` and `plain_id` in the CSV?

I see that the least significant 3 hex digits of the reader_id seem to currespond to the plain_id, however what are the most significant digits for?

For example, the app reports "Unknown (0x363)" which I know is Piola M2. How do I add it to the mappings file?

What could be a convenient way to record information while traveling without too much hassle and then add it to the app?

phcoder, 2019-06-19:
reader_id: at offset 28 is transport type: Metro is 1 and so on (See RicaricaMiLookup). Low bits is station number. So here reader_id is 0x01000363.
plain_id matches GTFS. Just look in the GTFS. E.g.
wget https://www.amat-mi.it/it/downloads/data/gtfs/current/Export_OpenDataTPL_Current.zip
unzip Export_OpenDataTPL_Current.zip stops.txt
phcoder@penguin:/tmp/rm$ grep -i Piola stops.txt 
..
"PIOLA","","PIOLA","","45.481145001027","9.22591680786948","","","","","Europe/Rome","2"
So plain_id is PIOLA

phcoder, 2019-06-19:
About recording: I take pictures of station signs and scans at the same time then match them using timestamps

phcoder, 2019-06-21:
About jvmCommonMain: in an ideal world we wouldn't even need it. It works fine in my intellij idea community edition latest version. Which IDE do you use?

Depau, 2019-06-21:
I'm using Android Studio. I'll try with IDEA

markhub-ai, 2019-08-25:
Just a site note: actually the "ricaricami" cards does not work with metrodroid, but the problem relies on the chip card type (Calypso 1.0 - ISO14443 Type B).
Calypso is hacked from the very moment it was launched (both ATM and Ricaricami use Calypso protocol) but only ATM is readable, therefore Calypso is not a problem. 
ATM cards are ISO14443 Type A cards (supported). Therefore, Ricaricami cards are "unreadable", even with an external reader attached to the phone (ACR122U) as stated in the following issue. 
I wonder if a proxmark3 can read this "NfcBaguette" card type...

See https://github.com/metrodroid/metrodroid/issues/262

Depau, 2019-08-25:
Old cards (issued more than 2 years ago - source: a quick check at my card and my friends') are plain Mifare Classic cards with a Calypso dataset and patches by @phcoder implement reading them. This includes both old ATM (orange) cards and Ricaricami (green) cards.

As far as I can remember you can communicate with the new cards with a proxmark device but IDK if the protocol has been fully reverse engineered and/or implemented. I'm quite sure that you cannot communicate with them using the Android API but I might be wrong.

phcoder, 2019-08-25:
First generation of Calypso cards uses Innovation non-standard protocol. Most Calypso cards use ISO 14443-B which is readable by almost all NFC-enabled phones. The main problem for supporting new ATM/Ricaricami cards is that I have no samples. If you can help me with this, please contact me privately

Depau, 2019-08-26:
I'll have to go to the ATM point next month anyway, I can get you a Ricaricami card. I'll contact you around that time.

phcoder, 2019-10-14:
I was passing through Milan and managed to get both new Ricaricami and ATM cards. Interestingly my old RicaricaMi still works.
New cards seem to be Calypso v1 using Innovatron pre-B' but without contact interface common for Calypso cards. So far it's just a speculation as I can't read the card so far

jacopo-j, 2019-10-16:
> New cards seem to be Calypso v1 using Innovatron pre-B' but without contact interface common for Calypso cards. So far it's just a speculation as I can't read the card so far
You are correct. You can try and read the card by sending raw commands and using libnfc and [this](https://github.com/nfc-tools/libnfc/blob/master/examples/pn53x-tamashell-scripts/ReadNavigo.sh) as an example

jacopo-j, 2019-10-17:
The ACR reader should also do fine as it works with libnfc

KonssnoK, 2021-12-01:
anybody picked up this topic?

> I was passing through Milan and managed to get both new Ricaricami and ATM cards. Interestingly my old RicaricaMi still works. New cards seem to be Calypso v1 using Innovatron pre-B' but without [...]

francescolf, 2022-03-08:
Is anyone still working on this? I have an ATM (orange) card issued in 2016 and I'm getting an `UnauthorizedException` when I read the card.