metrodroid/metrodroid #540: Crypto assets transfer

simsim314, 2019-08-01:
Is it possible to write on the nfc chips of the existing smart cards some new data related to crypto wallets? For example to have a wallet address and some smart contract haskey that will allow crypto asset transfer from one smart card to another? 

This question is related to the question: is it possible not hack the money stored on the card but yet transfer the currency between people without being forced to buy/pay to the third party for transportation? On the other note maybe it's possible to simply write a new transfer system that will support all transportations as well as cryptocurrency, so people would be able to pay for the ride in local currency as well as with crypto.

phcoder, 2019-08-01:
Some cards have some spare space. MifareClassic may have unused sectors. You may be able to create additional applications on MifareDesfire. You may be able to add an application on ISO 14443-A/B cards. But doing so has a risk of card being considered fraudulent and be banned.
Transit companies ot their integrators are sovereign in what they accept as payment, so you can't add a card format or payment system without partnering with them.
Most important of all this has nothing to do with Metrodroid, and hence is off-topic here

micolous, 2019-08-17:
Closing and locking as this is off-topic.

closed 2019-08-17 by micolous