metrodroid/metrodroid #635: Octopus card Years [QUESTION]

WingCH, 2019-09-24:
Base on your document about [Octopus](https://github.com/metrodroid/metrodroid/wiki/Octopus)

> cards allow a negative balance of up to 50 HKD (previously 35 HKD). This results in the balance being encoded differently.

How do I know when the card was released by programming?

Thanks to your help!

micolous, 2019-09-28:
The answer is I don't actually know, but I cheat. :)

Metrodroid reads cards and then saves a copy of the (accessible) card data. It also adds some metadata about when that copy was made.  For Octopus, if this scan was created before 2017-10-01, we offset by 35 HKD.  If this scan was created on or after 2017-10-01, we offset by 50 HKD (but I noticed we actually got this wrong, and this will be fixed by #638).

I assume (based on [Octopus FAQ](https://www.octopus.com.hk/en/consumer/customer-service/faq/get-your-octopus/about-octopus.html#3532)) that this change was handled by rewriting the balance field next time the card is used with an Octopus ticket validator.

I prioritise getting this right for folks who are frequently using their Octopus card -- at worst, the balance will be shown 15 HKD _lower_ until the next time they use it in the Octopus system.

micolous, 2019-10-05:
I have updated and included this explanation in the Wiki now: https://github.com/metrodroid/metrodroid/wiki/Octopus

closed 2020-12-31 by micolous