metrodroid/metrodroid #660: [MFC] Add support for Tehran Ezpay cards

HrBDev, 2019-09-30:
There's a lot of cards in the world, and asking for a card by name isn't good enough to get it supported.

**We do not record expressions of interest here.**

Please read https://github.com/metrodroid/metrodroid/wiki/New-cards before continuing.

## New card checklist

- [x] I have read and acknowledged the New cards wiki page.
- [x] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [x] I have attempted to understand the card format on my own.
- [x] I have provided notes on what I've found so far.

## Please answer all questions listed here:

- Name of the card: Tehran Ezpay
- Name of the transit agency: Tehran Urban and Suburban Railway Operations Co.
- Transit agency's website: http://metro.tehran.ir/
- Location(s): Tehran,Tehran,Iran
- Card media: MIFARE Classic

Is there any publicly accessible documentation for this card format?  If so, provide links:
No
Is the card similar to one we already support?
Don't know
Is there an official app for using the card?
No
How many sectors does this card have?
16
Did the card use any "default" key that comes with Metrodroid?
Only for sector 0
Does the card use diversified keys, static keys, or a combination?
Same keys for all cards
After installing proper keys, how many sectors could you read with Metrodroid (ie: not "unauthorized")?
All of them
With the data that you can read from the card, does it change after performing a transaction?
Yes

micolous, 2019-10-05:
Thanks!

Do you have some notes and annotated dumps on what you've found so far on this card?

If you can email some card dumps to myself and @phcoder, we should be able to check over things and tell you if it similar to another card we know.

Also, is there some open data (eg: GTFS, GeoJSON, KML, SHP, etc.) which describes the positions of stops on this network?

HrBDev, 2019-10-05:
unfortunately, I don't have your email.
I tried emailing the support email of metrodroid before opening this issue but it seems that they get deleted.

micolous, 2019-10-06:
Yup, sorry, that's to deal with spammers. 😞 

My email is: (my github username) at gmail.com (and is also listed in nearly every Metrodroid source file).

HrBDev, 2019-10-06:
> Also, is there some open data (eg: GTFS, GeoJSON, KML, SHP, etc.) which describes the positions of stops on this network?
https://wiki.openstreetmap.org/wiki/Tehran/Tehran_Metro

phcoder, 2019-10-06:
My email is also my nickname at gmail.com. Generally git log is the best way to get contributor's email