metrodroid/metrodroid #681: Seattle Monorail recognized as water taxi

frozenpandaman, 2019-10-08:
[image]

- Name of the card: ORCA
- Name of the transit agency: Seattle Monorail
- Location: Seattle, WA
- Metrodroid version: 2.9.43 (3097)
- Type: Line
- Mode of transport: Monorail

micolous, 2019-10-20:
Thanks for the report.

Unfortunately, the raw transaction field I'd need to look at to fix this isn't exposed in Metrodroid... so I'll need a copy of this card dump to fix it (... ▶️ `Advanced Info` ▶️  `...` ▶️  `Export` ▶️ `Share` / `Save to file`).

Please _don't_ attach it to this issue -- please email it to me: (my Github username at gmail.com).

Have you ever used the [King County Water Taxi](https://en.wikipedia.org/wiki/King_County_Water_Taxi) and have some _actual_ trips from it to compare?  It looks like there is some overlap with IDs for King County agencies.

frozenpandaman, 2019-10-20:
>I'll need a copy of this card dump to fix it

Great timing – the two monorail trips I have on my card are #s 9 and 10, i.e. just about to be overridden the next couple times I ride transit, as ORCA only stores data of the ten most recent trips on cards. I'll email you this shortly. 😄

>Have you ever used the King County Water Taxi and have some actual trips from it to compare? 

Not in the recent past nor do I do this regularly, so this would be hard for me to obtain. I could reach out to some friends or fellow Seattle transit enthusiasts to see if anyone could provide this info.

micolous, 2019-10-20:
Thanks!

If you've scanned it in Metrodroid, it'll be saved on your device and accessible in `History`, if you haven't explicitly deleted old scans.  Each of them is a point-in-time snapshot.

frozenpandaman, 2019-10-20:
Ahh, didn't realize all that information was saved/accessible via History as well. Handy! Emailed you the exported JSON file.

micolous, 2019-10-24:
It appears in both trips that you took (with different stations), they have the same Coach/Station ID (3).  [According to the Seattle Monorail website](https://www.seattlemonorail.com/buy-tickets/), it looks like they issue you a paper ticket after scanning your ORCA -- so they're doing simple purse transactions (and it doesn't look possible to figure out where you boarded).

Bits 288-308 appear to have another field in there that we don't yet decode, and some bits do change between your two Monorail trips.  It's not clear to me what the meaning of that is, and field doesn't appear to be set on other ORCA dumps I have, and doesn't appear to be set for your Link Light Rail journeys.

Unfortunately, I don't know [the full context behind the Water Taxi change](https://github.com/codebutler/farebot/pull/170) to be sure if this doesn't cause problems for Water Taxis.  For now, I'm treating things that looked like the previous Water Taxi type as "purse debit", then:

* Coach ID = 3: Seattle Monorail
* everything else: Water Taxi

Coach ID will continue to be shown for Water Taxi trips, so we should be able to sort that out.

frozenpandaman, 2019-10-25:
Thanks! During peak times (in the summer) Seattle Monorail operates two concurrent vehicles, the Red and the Blue monorail; during this time of year (and on the trips I took) only the Red one is in service. I'm not sure if these would have different coach IDs, or if they'd be entered as such on the machines used by staff to scan ORCA cards, but just an FYI.

> According to the Seattle Monorail website, it looks like they issue you a paper ticket after scanning your ORCA

FWIW, they don't do this (in my experience, or for anybody I know who has ridden it recently either). They just scan your card and let you through to board – that's it.

For me, it's not super important to have information where I boarded, compared to the app showing the vehicle type correctly (i.e. having this not display as the water taxi)… which sounds like it's fixable, so yay. :D

micolous, 2019-10-26:
> > According to the Seattle Monorail website, it looks like they issue you a paper ticket after scanning your ORCA
> 
> **EDIT**: I don't see where it says this on the page you linked? The text (unless I missed something) nor the video shows/indicates people getting paper tickets/receipts of any sort (by default) for trips paid with ORCA.

Ah, I am mistaken -- I thought I saw a paper ticket in the video.

Is the cashier actually on the vehicle, or are there cashiers at the stations?  It's hard to tell from that video.

frozenpandaman, 2019-10-26:
In both stations, a cashier sits in a booth (taking money, scanning cards, etc.) which you have to pass by in order to actually get up close to the vehicle & the boarding doors. The only staff member who rides the monorail is the operator/driver.

micolous, 2019-11-15:
Thanks!

@phcoder went to Seattle and took a Water Taxi – it looks like it _also_ uses Vehicle ID 3.  There is probably some other field that we're missing here ([probably in bits 288-308](https://github.com/metrodroid/metrodroid/issues/681#issuecomment-545847558)).

In 3.0.0, #688 means we'll recognise Water Taxi trips as Monorail trips, so this isn't really fixed properly yet. 😞

cookieguru, 2019-11-25:
The Water Taxi, Monorail, and streetside human [loaders](https://seattletransitblog.com/2012/08/02/metro-to-platoon-buses-use-loaders-in-dstt/) all use the same [handheld readers](https://www.flickr.com/photos/viriyincy/3561731244/) (or [PFTP](https://www.flickr.com/photos/viriyincy/6109994154/) in ORCA parlance).  Of all the ORCA agencies, KCM is the only one I know of that actually deploys these.  The Monorail is not part of KCM, however I've noticed the 2 handhelds they have each have KCM stickers on them.

Every single one shows up as KCM vehicle #3, regardless of which mode you're boarding.  It's not going to be possible to show a vehicle number on all three situations since the vehicle doesn't tag the card, a human with a reader does.

---

> Is the [Monorail] cashier actually on the vehicle, or are there cashiers at the stations? It's hard to tell from that video.

Both, depending on the time of day.  Late at night the cashier goes home so the train driver sells tickets and tags cards.

And feel free to tag me next time something ORCA-related comes up :)

micolous, 2019-12-01:
Thanks @cookieguru. That change to Farebot seems like a reasonable compromise.

cookieguru, 2019-12-01:
@micolous Is there a way to inspect bits 288-308 in a dump?  I can take a look at my scans to see if there's a pattern

micolous, 2019-12-16:
You want to look at File 0x2; each record is 48 bytes long, and they're concatenated.  Counting bytes from 0, that field is from byte 38 (288÷8) onwards in each record.

PR #688 (2019-11-02): orca: Fix Seattle Monorail, show raw data flelds
Fixes #681 and #686.
commit 83fd0ed0 (2019-11-02): orca: Add support for Seattle Monorail (fixes #681)