metrodroid/metrodroid #707: New Mexico City Card ISO [7816]

DevilIris, 2019-11-15:
There's a lot of cards in the world, and asking for a card by name isn't good enough to get it supported.

**We do not record expressions of interest here.**

Please read https://github.com/metrodroid/metrodroid/wiki/New-cards before continuing.

## New card checklist

- [ Yes] I have read and acknowledged the New cards wiki page.
- [ Yes] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [Yes ] I have attempted to understand the card format on my own.
- [ Yes] I have provided notes on what I've found so far.

### Please answer all questions listed here:

- Name of the card:  Movilidad Integrada (Integrated Mobility)
- Name of the transit agency: Sistema de Transporte Colectivo (STC),
- Transit agency's website: https://www.metro.cdmx.gob.mx
- Location(s):Mexico City
- Card media: ISO 7816(Calypso)

Is there any publicly accessible documentation for this card format?  If so, provide links:
I try to read the card with the app. And the app read it. i export the raw of 3 different cards with same version. This cards are on the .zip 
[Calypso Card MXC.zip](https://github.com/metrodroid/metrodroid/files/3850408/Calypso.Card.MXC.zip)
[SeminarioTransporte_FONADIN_FIMPE.pdf]

A Private Contract document about what terminal and systems they use to build to the Subways
(https://github.com/metrodroid/metrodroid/files/3850428/SeminarioTransporte_FONADIN_FIMPE.pdf)

[Contrato_Midsa_L3-p2.pdf](https://github.com/metrodroid/metrodroid/files/3850429/Contrato_Midsa_L3-p2.pdf)

Is the card similar to one we already support? (if you don't know, that's OK)
Maybe a French Card

Is there an official app for using the card?
No

What are the AIDs used on the card?
The AIDs are all in the export json. That i attached on the .zip

micolous, 2019-11-16:
I had a quick look over this, and this looks like Calypso (especially from the presentation you've linked).  Thankfully it appears that the Calypso dumper has pulled a bunch of data from your cards :)

Bits 4-16 of the Ticket Environment are `0x484`, the ISO 3166-1 numeric code for Mexico.  This part seems similar to `PisaTransitData` (Italy) and `RavKavTransitData` (Israel).  The offsets they use are _slightly_ different but it's effectively checking the same bits.

You won't get anything out of the parser at the moment, as there's no "fallback" behaviour for Calypso parsing.

_Semi-relatedly:_ `SeminarioTransporte_FONADIN_FIMPE.pdf` has _a lot_ of information about the state of transit cards in Mexico.  If I'm reading this correctly, they're trying to make a case for a national interoperable ticketing system across all of Mexico?

DevilIris, 2019-11-16:
About
> SeminarioTransporte_FONADIN_FIMPE.pdf
This card unify all the transports in Mexico City, also this card is the base for all the Transport Cards in Mexico.
All the Mexican cards works with calypso.
And the other document. Is the private contract of a MetroBus Line.
Sorry if the document is in spanish. If you need help on some translation tell me pls.