metrodroid/metrodroid #710: [iOS] Clipper card throws CardLostException

chrisballinger, 2019-12-05:
## Description of the issue

When scanning my primary Clipper card, it takes a long time and multiple re-positioning to get the card to register. After a while it scans to 100%, but then shows an info screen with a red banner with a "CardLostException" which seems to indicate that the read was interrupted. I tried scanning ~10 times and received the same result each time.

I'm not sure how to properly redact the raw logs so I'll refrain from posting those here.

The card itself still works fine on BART and AC Transit, although I have noticed it requires multiple scans sometimes so perhaps the card is starting to crap out (from the raw logs it looks like it was manufactured in 2011).

## Details from About screen:

- Metrodroid version: HEAD 4fdc09a523c4fbf9a315a2d94601e783aff0e64b
- Device Model: iPhone 11 Pro Max
- Android/iOS version: 13.2.3

## Checklist

- [x] The card appears in the "supported cards" list in Metrodroid on my device.
- [x] I have copied all the details from the about screen into this bug.
- [x] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.

micolous, 2019-12-07:
If you want to send through logs, you can email them to me (my github username at gmail dot com).

Card faults generally result in "tag lost", and then iOS shuts down the NFC session without recourse.  I'd be interested to see if you see the same issue with an Android device: Clipper cards are all DESFire so should be near-universally readable.  

Fareboxes generally have better antennas and more direct control over the NFC controller (so can apply some retry logic); if it's not reliable there either you're stuck and it's probably time to replace it.

I've seen plenty of older cards that don't work with some phones, because the antenna performance on both the card and the device is marginal.  It entirely depends on its tuning (out of our control).

With some devices it's hard to find the "sweet spot" for reading NFC -- for the iPhone 7 I use as a test device, the best spot at the top edge of the phone.  Cards with more data, and slow cards are more vulnerable to issues; but [my Clipper (manufactured in 2017) takes about 2 seconds to read](https://youtu.be/78-Th47Sgfk), and there's not a huge amount of data exchanged.

chrisballinger, 2019-12-07:
Your video was super helpful, and now I've got it working 100% of the time, even with the older cards. This issue can probably be filed under PEBKAC.

 Thank you!

closed 2019-12-07 by chrisballinger