metrodroid/metrodroid #718: [BUG] Amiibo not recognized

ranma, 2019-12-29:
## Description of the issue
Amiibo not recognized
Green Yoshi NVL-201 is only recognized as raw data.

## For card-specific issues (delete if not applicable)

- Name of the card type: Amiibo
- Name of the transit agency: Nintendo
- Location: 

Please export a card dump from Metrodroid.

Please do not attach it to this issue -- you may be asked for it later via private email.

## Steps to reproduce

1.
2.
3.

## Expected behaviour
Yoshi recognized as Amiibo

## Actual behaviour
Yoshi recognized as raw data

## Workarounds (if any)

## Device Logs

Android: Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

iOS: Generate a sysdiagnose using the instructions here: https://www.jessesquires.com/blog/how-to-sysdiagnose-ios/

Please do not attach it to this issue -- you may be asked for it later via private email.

## Details from About screen:

- Metrodroid version: 3.0.0 (3920)
- Device Model: Pixel 3A (sargo)
- (Android only) Manufacturer: Google
- Android/iOS version: 10 (QQ1A.200105.002)
- (Android only) NFC:
- (Android only) MIFARE Classic:

## Checklist

- [ x] The card appears in the "supported cards" list in Metrodroid on my device.
- [ x] I have copied all the details from the about screen into this bug.
- [x ] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [x ] I have generated an Android bug report ZIP, and can email it privately if needed.

micolous, 2020-01-14:
This sounds like a data problem.  Amiibos are kinda strange, but we treat them like transit cards, and this kind of issue is what we'd call "missing data".

NVL-201 appears to be the Yarn series, which we've (supposedly) already got data for: https://github.com/N3evin/AmiiboAPI/blob/b5703ef93413941aa58c46ef42202af6c24c160d/database/amiibo.json#L161-L169

Provided you've not stored any game data with the Amiibo, please use that raw data view to save a copy of the data, and attach it to this bug.

If you have data associated with your Amiibo, or would otherwise keep it private, you can email it to (my github username) at gmail.com.

I'm unable to take further action without a dump of this Amiibo.

ranma, 2020-01-15:
{
    "tagId": "04138f9a844981",
    "scannedAt": {
        "timeInMillis": 1577641153202,
        "tz": "Europe/Berlin"
    },
    "mifareUltralight": {
        "cardModel": "NTAG215",
        "pages": [
            {
                "data": "04138f10"
            },
            {
                "data": "9a844981"
            },
            {
                "data": "d6480fe0"
            },
            {
                "data": "f110ffee"
            },
            {
                "data": "00000000"
            },
            {
                "data": "6175aba1"
            },
            {
                "data": "6bf590a2"
            },
            {
                "data": "eeb762e9"
            },
            {
                "data": "fe8f6b67"
            },
            {
                "data": "fd890fbf"
            },
            {
                "data": "296cdf77"
            },
            {
                "data": "87348b7e"
            },
            {
                "data": "0f830ebe"
            },
            {
                "data": "54e51589"
            },
            {
                "data": "61c6ff27"
            },
            {
                "data": "3b27b480"
            },
            {
                "data": "a5310dca"
            },
            {
                "data": "b7f67689"
            },
            {
                "data": "f74c72af"
            },
            {
                "data": "f978be7d"
            },
            {
[... 322 lines cut ...]
            },
            {
                "data": "d8b1055b"
            },
            {
                "data": "4e203995"
            },
            {
                "data": "01000fbd"
            },
            {
                "data": "00000004"
            },
            {
                "data": "5f000000"
            },
            {
                "data": "00000000"
            },
            {
                "data": "00000000"
            }
        ]
    }
}

ranma, 2020-01-15:
Notably it looks like the 0x0003010200410302 sequence does appear inside the raw data.

micolous, 2020-03-15:
Page 4 looks different to what Metrodroid expects: it [should](https://www.3dbrew.org/wiki/Amiibo#Page_layout) start with `a5`, your Yoshi has `00000000`.  As a result, Metrodroid doesn't think that this is an Amiibo.

After applying the change in #733, this is recognised as an Amiibo:

```
% ./metrodroid-cli parse ~/dev/cards/amiibo/green-yoshi-nvl-201.json
card UID = <04138f9a844981>
   name = Amiibo
   serial = null
   info
      Type: Yarn
      Character: Yoshi [0x3]
      Character variant: 1
      Model number: 65
      Series: Yoshi's Woolly World
```

I think this is because this Amiibo has never been written to by Nintendo hardware / software, is that correct?

ranma, 2020-03-22:
Could be, but I'm not fully sure. Thanks for the investigation!

commit 6a9491bb (2020-08-24): Fix #718: Support Amiibos that have never been written to
commit ef1e3444 (2020-08-24): Merge pull request #733 from micolous/fix-718

Fix #718: Support Amiibos that have never been written to
closed 2020-08-24 by micolous