metrodroid/metrodroid #723: [ENHANCEMENT] Mifare Plus 2k support

AetherEternity, 2020-02-05:
Tried to use this app (v3.0.0 release) with my Troika, but ran into lack of mifare plus 2k (32 standart sectors) support.

Relevant information:
ATS 0C0C757780 | 02C1052F2F
ATQA  00  44  
SAK  08

micolous, 2020-03-15:
As far as I'm aware, we _already_ support reading MIFARE Classic (MFC) 1K, 2K and 4K on Android, and [we get that information][1] from [the Android platform][2].

Your PR, #724, fixes converting (raw) dumps from other tools, but that doesn't get used on Android.  But thanks for clearing that up. :)

@phcoder added support for reading MIFARE Plus (MFP) cards in SL0 and SL3 modes (AES) some time ago.  When reading in this mode, [we check for the presence of sector 32][5], to determine whether it is a 32 (2K) or 40 (4K) sector card.

Looking at your card:

* The `ATQA` [is used by both MIFARE Ultralight (all variants), MIFARE Classic 1K (with 7-byte UID), and MIFARE Plus 2K (both EV1 and non-EV1, with 7 byte UID in any security mode)][3].  

* The `SAK` [looks like a MIFARE Plus in _SL1_ mode _or_ MIFARE Classic][3].

To use the MIFARE Plus SL0/SL3 reader, [we require SAK=0x20][6], as [this is the expected configuration][3].

When a MIFARE Plus is 2K in SL1 mode, the [it acts as a MIFARE Classic 1K][7]:

> [image]

But [the data sheet][7] also indicates "the 2kB EEPROM memory ... is organized in 32 sectors of 4 blocks" and Figure 3 shows `CRYPTO1 key` in sector 31 block 3 (which would only exist on a MFC 2K).

Given that there's no MFP 1K, I suspect that the disparity is a result of relative unpopularity of MFC 2K compared to MFC 1K, and the cost difference between 1K and 2K EEPROMs when the MFP series was designed.

Even if those sectors _were_ in use, this _is not_ required to read a **Troika**: [our reader code only uses sectors 1, 4, 7, and 8][4], which are _all_ on a 1K card.

tl;dr: I think everything here is _working as intended_:

* we can already read MFP 2K cards when in SL0 or SL3 modes
* we can already read MFP cards in SL1 mode by treating them as MFC
* MFP 2K in SL1 mode acts like a MFC 1K
* MFP 4K in SL1 mode acts like a MFC 4K
* Troika cards (including hybrids) only use sectors present on the MFC 1K

[1]: https://github.com/metrodroid/metrodroid/blob/7110bbbb78cfb5f3fc0f4224fd1b7168bcb6736b/src/main/java/au/id/micolous/metrodroid/card/classic/ClassicCardTechAndroid.kt#L48
[2]: https://developer.android.com/reference/android/nfc/tech/MifareClassic#getSectorCount()
[3]: https://www.nxp.com/docs/en/application-note/AN10833.pdf
[4]: https://github.com/metrodroid/metrodroid/blob/53d7e4a6b017e87ee07a6de485212e9c1e440f0d/src/commonMain/kotlin/au/id/micolous/metrodroid/transit/troika/TroikaTransitData.kt#L57-L62
[5]: https://github.com/metrodroid/metrodroid/blob/dad2893a04c5a6025dbe32be99f54a68d755893c/src/commonMain/kotlin/au/id/micolous/metrodroid/card/classic/PlusProtocol.kt#L133
[6]: https://github.com/metrodroid/metrodroid/blob/e7e8a42dffcd230378568577a96cdc4d590ce1b5/src/commonMain/kotlin/au/id/micolous/metrodroid/card/classic/ClassicReader.kt#L170
[7]: https://www.nxp.com/docs/en/data-sheet/MF1SPLUSX0Y1_SDS.pdf

phcoder, 2020-10-11:
I agree with the assessment. MFP 2K in SL1 mode behaves like MFC 1K and the other half of the sectors are inaccessible by design. So this should be closed as invalid

closed 2020-12-31 by micolous

AtunCL, 2023-04-12:
@micolous i need help with tool proxmark3 specifically for mifare 2k sl0/sl3 keys and dumps