metrodroid/metrodroid #725: [BUG] Scanning Clipper cart causes ClassCastException

rtyler, 2020-02-08:
## Description of the issue

When scanning a Clipper Card, I'm getting a:

```
ClassCastException: au.id.micolous.metrodroid.transit.clipper.ClipperRefill cannot be cast to java.lang.Comparable
```

It's unclear to me how to generate a full proper stacktrace.

## For card-specific issues (delete if not applicable)

- Name of the card type: Clipper
- Name of the transit agency: Clipper
- Location: SF

Please export a card dump from Metrodroid.

Please do not attach it to this issue -- you may be asked for it later via private email.

## Steps to reproduce

1.
2.
3.

## Expected behaviour

Not sure tbh :)

## Actual behaviour

Raw data visible, but the red exception bar at the bottom.

## Workarounds (if any)

## Device Logs

Android: Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

iOS: Generate a sysdiagnose using the instructions here: https://www.jessesquires.com/blog/how-to-sysdiagnose-ios/

Please do not attach it to this issue -- you may be asked for it later via private email.

## Details from About screen:

- Metrodroid version: 2.9.34 (570)
- Device Model: Galaxy S4
- (Android only) Manufacturer:
- Android/iOS version: 9 (Lineage OS)
- (Android only) NFC:
- (Android only) MIFARE Classic:

## Checklist

- [x] The card appears in the "supported cards" list in Metrodroid on my device.
- [ ] I have copied all the details from the about screen into this bug.
- [ ] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [ ] I have generated an Android bug report ZIP, and can email it privately if needed.

phcoder, 2020-02-08:
This is a very old bug and a very old metrodroid version. Please upgrade

rtyler, 2020-02-09:
@phcoder ah interesting, I just grabbed the version listed in F-Droid, guess I'll have to build my own APK :man_shrugging:

closed 2020-02-09 by rtyler