metrodroid/metrodroid #730: Distributing Metrodroid on iOS

robbiet480, 2020-02-20:
Hi,

I've been following the development of Metrodroid for iOS closely over the last few months and have been actively using it on my device. I think it's at a point where it can be safely submitted to the App Store. I thought I'd reach out to offer to do so under my pre-existing Apple Developer Account. I am the author (and publisher) of one of the largest open source iOS apps, [Home Assistant](https://apps.apple.com/us/app/home-assistant/id1099568401?ls=1&cache=454545) which has over 200k downloads. I have almost 4 years of experience on all things iOS and App Store. I also was the author of [TransitPal](https://github.com/robbiet480/TransitPal), a short lived project to build a native app for scanning transit cards that used MdST. I'm happy to distribute Metrodroid under my account, assuming @micolous doesn't wish to pay $99/year for the privilege. 

Here's the actions that I believe need to be taken to make this possible:

- Generate App Store metadata: things like description and keywords. Most of this can be reused from Google Play. Please note that none of the metadata can reference or acknowledge Android, Google, Play Store, etc. In the app is fine, just not outside.
- I will configure the app with a personal bundle ID, for example `io.robbie.metrodroid`. This way there won't be a conflict if others want to request certificates or publish under the `au.id.micolous.metrodroid.phcoder` bundle ID they can.
- I will provide access to my App Store account to at least one person, probably @micolous.
- Determine licensing issues if any. I am aware of #703, would be good to get that merged/approved by relevant parties

In addition to (eventually) submitting to the store, i'm happy to quickly offer Metrodroid for testing via TestFlight. We can also obviously hook in Travis to automatically build and release every commit to TestFlight.

Anyway, let me know how I can help or what information I can provide, i'm happy to do so and hope to see Metrodroid on the App Store soon!

cc @phcoder

robbiet480, 2020-02-20:
Testflight is now live at https://testflight.apple.com/join/O7OYdFao

phcoder, 2020-02-20:
Please don't distribute it on app store until we resolve the license conflict with apple Eula. Both Michael and me are currently working on it

robbiet480, 2020-02-20:
I don't plan to distribute on the store without both of your full approval, don't worry. I hope that TestFlight is okay for now, but if not I can delete the link.

micolous, 2020-02-24:
Heya, thanks for your enthusiasm, and apologies for not communicating on this.  It's been a busy few months for me, and I've been pre-occupied with real-life issues and fixing up some of my various other projects.

_Regarding the fees,_ both @phcoder and I have Apple developer accounts paid up already. [This is required for us to deploy iOS applications that used NFC APIs to our _own_ devices](https://help.apple.com/developer-account/#/dev21218dfd6).

_Regarding the icon_, I built it purely in raster form. [The source SVG used for the logo](https://github.com/metrodroid/metrodroid/blob/master/extra/logo.svg) is based on GTFS-sourced geospatial data, and even _after_ significantly simplifying its geometry, it was still extremely intricate and too difficult to work with in vector form.

I think this may need an update for Android because of Google Play Store policy changes.

_Regarding #703,_ [according to the Free Software Foundation, Apple's App Store EULA is incompatible with the GPLv2](https://www.fsf.org/blogs/licensing/more-about-the-app-store-gpl-enforcement) (which doesn't contain [anti-tivoization clauses added in GPLv3](https://en.wikipedia.org/wiki/Tivoization#GPLv3)).  I discovered this issue when I was trying to set up things in App Store Connect and TestFlight.

I consider an open source project's license a tool to encode the wishes of people who contributed to it.  I want to ensure that I respect the wishes of those who have contributed to FareBot and Metrodroid over the years.  As a result, I decided to stop further iOS work (even setting up TestFlight) _until_ that issue can be resolved.

@phcoder has gone through `git` logs and made a list of contributors we need to get in contact with to get that merged, and I started drafting a form letter for this.  We already have the "OK" from the folks who have contributed the majority of lines-of-code, but we need to get _all_ of them onboard to go ahead with _any_ distribution of Metrodroid binaries for iOS (or start removing their contributions).

Right now that's the main blocker for iOS deployment; and this is before we start going through the App Store approvals process.

robbiet480, 2020-02-24:
Great, thanks for the thorough update @micolous. Had I known any of this, I wouldn't have moved forward at all with publishing on my own. I'll go ahead and disable my TestFlight link for now as it seems like you or @phcoder will be able to get this published on your own. 

Let me know if I can be of help in any way to get this on the App Store.

micolous, 2020-02-24:
Thanks!

Also, thanks for your PR #731, that's still helpful because we'd have eventually encountered these problems :)

robbiet480, 2020-02-24:
Glad that helps a bit.

I'll also point out that the app sailed through TestFlight beta review without even a question from Apple. To be clear, there's a much lower barrier to pass that but its a good sign at least that they didn't instantly reject it for a violation or because they were unable to properly test.

micolous, 2020-03-29:
I've written up a form letter a couple of weeks ago, but I haven't heard from @phcoder recently – I suspect he's quite busy with real life commitments now.

If you don't mind, I've shared a copy of what I wrote with you for review (Google Doc).  After that, I can start contacting people.

robbiet480, 2020-03-29:
The only suggestion I have for the letter is you probably need a Google Form linked from it to accept contributors approvals, right? Or are you hoping to just have them reply to the email and you'll manually manage it.

Ivan0x17, 2025-06-09:
> Testflight is now live at https://testflight.apple.com/join/O7OYdFao

It's possible join to beta now?

Thank you so much in advance for the reply