metrodroid/metrodroid #735: [MFC] Troika, support for layout 0xE; subLayout 0x5.

monster1025, 2020-05-07:
Hello. Thanks for such great troika contents description.

I have tried to write python parser for troika cards and discovered that some of my cards have layout 0xE and new subLayout 0x5. 
By opening this issue I hope we can decode it too.
My sample python code is bellow.

## New card checklist
- [x] I have read and acknowledged the New cards wiki page.
- [x] I can provide dumps, transaction history, and balance records for more than one card from the operator. (Due to quarantine I can't buy new cards now, but I can do it later).
- [x] I have attempted to understand the card format on my own.
- [x] I have provided notes on what I've found so far.

## Please answer all questions listed here:
- Name of the card: Troika (Тройка)
- Name of the transit agency: Moscow Metro
- Transit agency's website: https://mosmetro.ru
- Location(s): Russia, Moscow
- Card media: MIFARE Classic

**Is there any publicly accessible documentation for this card format?  If so, provide links:**
https://www.mos.ru/authority/documents/doc/22526220/

**Is the card similar to one we already support?**
Maybe SubLayout 0x3

**Is there an official app for using the card?**
Мой проездной (https://www.vtb.ru/personal/online-servisy/moy-proezdnoy/), but it decodes data via web-service query.

**How many sectors does this card have?**
16 sectors/64 blocks.

**Did the card use any "default" key that comes with Metrodroid?**
Yes.

**Does the card use diversified keys, static keys, or a combination?**
Static keys

**After installing proper keys, how many sectors could you read with Metrodroid (ie: not "unauthorized")?** 
I can read all sectors data.

**With the data that you can read from the card, does it change after performing a transaction?**
Sure it changes, but I can't check right now.

Here is my sample python script with 2 dumps of sector 8 troika card. First one is "new format" layout 0xE; subLayout 0x5 and second one is "old format" layout 0xE; subLayout 0x3.
As you can see from script output that preamble and ticket type are correct, but all ticket data have different format.
Hope anybody from Moscow can provide some other dumps.

Output for new format by old parser code:
```
magic: 0x45d
ticketType: 0xb1
ticketType2: 0x5db1
ticketNumber: 8601632
layout: 0xe
sublayout: 0x5
unknown: 0x521
balance: 0x157e00
balance dec: 1408512
expDate: 0x4722 (2041-11-09 00:00:00)
lastValidator: 0x13e6
lastValidationTime: 0x230000 (2020-05-11 21:20:00)
lastTransfer: 0x7
lastTransports: 0x3c0
fareCode: 0x0
hash: 0xe4d5c115
```

Python script:
```
import binascii
from datetime import datetime
from datetime import timedelta

# troika - new format (layout E/subLayout 5), balance 348 (rub).
data = '45DB100834020E2A3915213D6740280013E646000021FC055F80016EE4D5C11513E646000021FC055F80016EE4D5C115'
# troika - old format (layout E/subLayout 3), balance 0 (rub).
#data = '45DB101958FBCE19768AA400000000002C013D460A001400000010009BB56E632C013D460A001400000010009BB56E63'

binData = bin(int(data, 16))[2:].zfill(256)
print(binData)

magic = binData[0:12]
print("magic: {}".format(hex(int(magic, 2))))

ticketType = binData[12:20]
print("ticketType: {}".format(hex(int(ticketType, 2))))

ticketType2 = binData[4:20]
print("ticketType2: {}".format(hex(int(ticketType2, 2))))

ticketNumber = binData[20:52]
print("ticketNumber: {}".format(int(ticketNumber, 2)))

layout = binData[52:56]
print("layout: {}".format(hex(int(layout, 2))))

sublayout = binData[56:61]
print("sublayout: {}".format(hex(int(sublayout, 2))))

unknown = binData[77:88]
print("unknown: {}".format(hex(int(unknown, 2))))

balance = binData[188:210]
print("balance: {}".format(hex(int(balance, 2))))
print("balance dec: {}".format(int(balance, 2)))

expDate = binData[61:77]
expDateDate = datetime.strptime("1992-01-01", '%Y-%m-%d') + timedelta(days=int(expDate, 2))
print("expDate: {} ({})".format(hex(int(expDate, 2)), expDateDate))

lastValidator = binData[128:144]
print("lastValidator: {}".format(hex(int(lastValidator, 2))))

lastValidationTime = binData[144:167]
lastValidationTimeDate = datetime.strptime("2016-01-01", '%Y-%m-%d') + timedelta(minutes=int(lastValidationTime, 2))
print("lastValidationTime: {} ({})".format(hex(int(lastValidationTime, 2)), lastValidationTimeDate))

lastTransfer = binData[171:178]
print("lastTransfer: {}".format(hex(int(lastTransfer, 2))))

lastTransports = binData[178:188]
print("lastTransports: {}".format(hex(int(lastTransports, 2))))

fareCode = binData[210:212]
print("fareCode: {}".format(hex(int(fareCode, 2))))

hash = binData[224:256]
print("hash: {}".format(hex(int(hash, 2))))
```

monster1025, 2020-05-29:
Same card in two formats: old and new.

# troika - old format (layout E/subLayout 3), balance 0 (rub).
data = '45DB101958FBCE19768AA400000000002C013D460A001400000010009BB56E632C013D460A001400000010009BB56E63'

# troika - new format (layout E/subLayout 5), balance 50 (rub).
data = '45DB101958FBCE2A4915216AA1A008000D1C0A000004E20B004001001EB7ADDE0D1C0A000004E20B004001001EB7ADDE'

monster1025, 2020-06-06:
Balance (in kopeyka, 0.01rub) found in bits 170-186:

balance = binData[170:186]

micolous, 2020-07-05:
Thanks for the tips!  Unfortunately, I'm not greatly familiar with the Troika parsing code, and I can't read Russian. :(

@phcoder do these changes seem reasonable for you?

phcoder, 2020-10-11:
I made https://github.com/metrodroid/metrodroid/pull/739
Some fields are unclear to me. Can you see how the fields in debug tab change after different taps?
The layouts E.4 and E.6 are still missing. They're used by multi-ride passes

closed 2021-10-06 by phcoder