metrodroid/metrodroid #737: iOS build failing, docs might need to be more specific?

indirect, 2020-08-01:
The docs for iOS say the requirements are:

> macOS 10.14 Mojave or later

I'm running 10.15.

> Android SDK (for technical reasons)

```
❯ sdkmanager --version
26.1.1
```

> Xcode 11.0 or later

```
❯ xcodebuild -version
Xcode 11.6
Build version 11E708
```

> Java runtime environment

```
❯ java -version
openjdk version "1.8.0_262"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_262-b10)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.262-b10, mixed mode)
```

(I've also tried OpenJDK 11 and 13, as well as AdoptOpenJDK 9).

On the Xcode build step that runs gradle, I'm getting this error:

```
e: /Users/andre/src/metrodroid/metrodroid/src/commonMain/kotlin/au/id/micolous/metrodroid/transit/clipper/ClipperTrip.kt: (74, 53): Unresolved reference: Integer
```

<details>
<summary>
Here's the full log from running the command with --info:
</summary>

```
❯ ./gradlew :packForXCode -PXCODE_CONFIGURATION=Debug --info
Initialized native services in: /Users/andre/.gradle/native
Found daemon DaemonInfo{pid=49503, address=[1c10e494-fc27-4111-9182-d16ff75a29e0 port:61499, addresses:[/0:0:0:0:0:0:0:1, /127.0.0.1]], state=Idle, lastBusy=1596236604342, context=DefaultDaemonContext[uid=774bedb7-f923-4c96-b284-282987cf7bf7,javaHome=/usr/local/Cellar/openjdk@11/11.0.8/libexec/openjdk.jdk/Contents/Home,daemonRegistryDir=/Users/andre/.gradle/daemon,pid=49503,idleTimeout=10800000,priority=NORMAL,daemonOpts=--add-opens,java.base/java.util=ALL-UNNAMED,--add-opens,java.base/java.lang=ALL-UNNAMED,--add-opens,java.base/java.lang.invoke=ALL-UNNAMED,--add-opens,java.prefs/java.util.prefs=ALL-UNNAMED,-XX:MaxMetaspaceSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xms256m,-Xmx512m,-Dfile.encoding=UTF-8,-Duser.country=US,-Duser.language=en,-Duser.variant]} however its context does not match the desired criteria.
Java home is different.
Wanted: DefaultDaemonContext[uid=null,javaHome=/Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home,daemonRegistryDir=/Users/andre/.gradle/daemon,pid=56390,idleTimeout=null,priority=NORMAL,daemonOpts=-XX:MaxMetaspaceSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xms256m,-Xmx512m,-Dfile.encoding=UTF-8,-Duser.country=US,-Duser.language=en,-Duser.variant]
Actual: DefaultDaemonContext[uid=774bedb7-f923-4c96-b284-282987cf7bf7,javaHome=/usr/local/Cellar/openjdk@11/11.0.8/libexec/openjdk.jdk/Contents/Home,daemonRegistryDir=/Users/andre/.gradle/daemon,pid=49503,idleTimeout=10800000,priority=NORMAL,daemonOpts=--add-opens,java.base/java.util=ALL-UNNAMED,--add-opens,java.base/java.lang=ALL-UNNAMED,--add-opens,java.base/java.lang.invoke=ALL-UNNAMED,--add-opens,java.prefs/java.util.prefs=ALL-UNNAMED,-XX:MaxMetaspaceSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xms256m,-Xmx512m,-Dfile.encoding=UTF-8,-Duser.country=US,-Duser.language=en,-Duser.variant]

  Looking for a different daemon...
The client will now receive all logging from the daemon (pid: 56035). The daemon log file: /Users/andre/.gradle/daemon/5.4.1/daemon-56035.out.log
Starting 3rd build in daemon [uptime: 7 mins 30.041 secs, performance: 100%, GC rate: 0.25/s, heap usage: 7% of 358.1 MB, non-heap usage: 46% of 268.4 MB]
Using 8 worker leases.
Starting Build

> Configure project :buildSrc
Evaluating project ':buildSrc' using build file '/Users/andre/src/metrodroid/metrodroid/buildSrc/build.gradle.kts'.
file or directory '/Users/andre/src/metrodroid/metrodroid/buildSrc/src/main/resources', not found
file or directory '/Users/andre/src/metrodroid/metrodroid/buildSrc/src/main/groovy', not found
file or directory '/Users/andre/src/metrodroid/metrodroid/buildSrc/src/main/kotlin', not found
Selected primary task 'build' from project :
:buildSrc:compileKotlin (Thread[Execution worker for ':buildSrc',5,main]) started.

> Task :buildSrc:compileKotlin UP-TO-DATE
file or directory '/Users/andre/src/metrodroid/metrodroid/buildSrc/src/main/kotlin', not found
Caching disabled for task ':buildSrc:compileKotlin' because:
  Build cache is disabled
Skipping task ':buildSrc:compileKotlin' as it is up-to-date.
[... 537 lines cut ...]
> Task :linkMetrolibDebugFrameworkIOS FAILED
:linkMetrolibDebugFrameworkIOS (Thread[Execution worker for ':',5,main]) completed. Took 39.126 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':linkMetrolibDebugFrameworkIOS'.
> Process 'command '/Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home/bin/java'' finished with non-zero exit value 1

* Try:
Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 40s
10 actionable tasks: 1 executed, 9 up-to-date
```

</details>

Can somebody tell me what I'm doing wrong, or what's missing from the iOS building docs? Thanks!

micolous, 2020-08-16:
I think you have an old checkout of the code – I believe this issue was fixed in https://github.com/metrodroid/metrodroid/commit/0b46b9063d5a23e4dfde23575ebf28de54e0f560.

indirect, 2020-08-16:
I got this error from a checkout of 83add7ee6d22baa4c2f5f75ea0e4c9656b948a5c. As far as I can tell, that already includes 0b46b9063d5a23e4dfde23575ebf28de54e0f560.

```
❯ git log -1
commit 83add7ee6d22baa4c2f5f75ea0e4c9656b948a5c (HEAD -> master, origin/master, origin/HEAD)
Author: WaldiS <sto@tutanota.de>
Date:   Mon Jul 27 18:23:47 2020 +0000

    Translated using Weblate (Polish)
    
    Currently translated at 14.2% (3 of 21 strings)
```

indirect, 2020-08-16:
That said, I just tried doing everything again from scratch and... it worked? I didn't even `git pull` this time. Thanks for taking a look at my question!

closed 2020-08-16 by indirect