metrodroid/metrodroid #738: [BUG] Seattle ORCA history display broken

frozenpandaman, 2020-10-10:
ORCA card reads seem to be displayed incorrectly on the "History" tab:
[image]

## Card-specific information
- Name of the card: ORCA
- Name of the transit agency: Various (incl. King County Metro, Sound Transit, Seattle Monorail)
- Location: Seattle, WA

## Details from About screen

- Metrodroid version 3.0.0 (3920)
- Device Model: Pixel XL (marlin)
- Manufacturer: Google
- Android version: 8.1.0
- NFC: On
- MIFARE Classic: Supported

## Checklist

- [x] The card appears in the "supported cards" list in Metrodroid on my device.
- [x] I have copied all the details from the about screen into this bug.
- [x] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [x] I have generated an Android bug report ZIP, and can email it privately if needed.

micolous, 2020-12-31:
I don't understand what the "correct behaviour" is here, so I don't understand what you're referring to here as the problem, or if you're re-reporting #681 against a different version of Metrodroid.

The `agency = 0x4` values looks like you have "raw data" mode turned on, that's expected.

frozenpandaman, 2020-12-31:
@micolous No, that other bug is unrelated. Huh, I have no clue how that got enabled, didn't even know it was an option – but yes, changing "Raw level" to None in Developer options seems to have fixed this. Thanks!

closed 2020-12-31 by frozenpandaman