metrodroid/metrodroid #744: [BUG] A display issue

franklinbill, 2020-12-28:
## Description of the issue
Dear developer!
In the process of using, I found a bug in layout / display. As shown in the figure, in the first page, the text overlap each other. This may be an adaptation problem. I suspect it is because **display and font size** are set to large (mainly due to **display size**).
Because of some problems with my eyesight, I often use the largest font and display.
[image]
App version is 2.9.37, my mobile phone is Google nexus 6, screen resolution setting is 1920* 1080, **_font size and display size are the largest ! ! !_**, Android system is 8.0.
The above is a description of the problem. The overall experience of the software is good. You can see that the design is very attentive. Thank you very much for the efforts of the developers all the time. I hope to receive a reply.
[image]

## For card-specific issues (delete if not applicable)

- Name of the card type:
- Name of the transit agency:
- Location:

Please export a card dump from Metrodroid.

Please do not attach it to this issue -- you may be asked for it later via private email.

## Steps to reproduce

1.use the largest font and display
2.see the issue

## Expected behaviour
I hope that the page can be set to scroll, so that the problem caused by the display size of large can be solved.

## Actual behaviour
As shown in the figure, in the first page, the text overlap each other. 

## Workarounds (if any)

## Device Logs

Android: Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

iOS: Generate a sysdiagnose using the instructions here: https://www.jessesquires.com/blog/how-to-sysdiagnose-ios/

Please do not attach it to this issue -- you may be asked for it later via private email.

## Details from About screen:

- Metrodroid version: 2.9.37
- Device Model:  Google nexus 6
- (Android only) Manufacturer:
- Android/iOS version: 8.0
- (Android only) NFC:
- (Android only) MIFARE Classic:

## Checklist

- [√] The card appears in the "supported cards" list in Metrodroid on my device.
- [√] I have copied all the details from the about screen into this bug.
- [√] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [√] I have generated an Android bug report ZIP, and can email it privately if needed.

micolous, 2020-12-29:
Oops, that's definitely not intended.  There's no scrolling on that view at all, so it can't handle having the font size "too large".

Are there any views other than the About screen which you see this problem on?

micolous, 2020-12-31:
I ran the current development version of Metrodroid in an emulator (emulating Pixel 3), with font and display size set to largest, and had a look:

## Views that look OK:

* Main screen
* Supported cards
* History
* Card view:
   * Ticket with name, balance and expiry
   * Pass with passenger count, status, and expiry
   * Trip history entries with agency, start and end stops, start and end times, and price
   * Info
* Advanced card info raw/manufacturing info views
   * MFC
   * MFD
   * FeliCa
   * Unsupported card banner

## Views that have problems:

* About screen (overlapping elements, as above)

* Trip map (map labels do not appear to be scaled – but unsure what we can do about this)

   [image]

* Advanced card info: title and subtitle can get cut off; but unsure how fixable that is. 

   [image]

* Preferences: some of the option names in Developer Options and for Map settings are cut off; but these are also explained in the description below it.

   [image]

   It looks like the Settings app has properly word wrapped preferences entries, so it should be possible to fix this one with some tweaks.

franklinbill, 2020-12-31:
Thanks*100000 for your reply! It's not bad on other pages. I agree with your analysis of the problem. Looking forward to your new version.

micolous, 2020-12-31:
Progress so far on staged changes (not pushed live):

## Use multi-line preference titles everywhere:

Screenshot at largest size; now long settings titles aren't cut off anymore.

[image]

This was also a problem when running Metrodroid in German at default size, which also gets fixed up by this change:

 [image]

## About screen fixes

At largest size, the view scrolls (scroll bar shows up when you touch to scroll):

[image]

At default size, elements are still spaced out as they were before:

[image]

micolous, 2020-12-31:
Unfortunately, it's very difficult to fix the app title bars used on the `Map` and `Advanced Card Info` views, because they're part of `ActionBar` (in the Android platform and AppCompat Support Library), which forces all titles and subtitles to be ellipsized, with no way to change it.

The only way around this would be to use custom display logic, which is extremely tedious to work with, and risks introducing other problems.  It'd be nice if the AppCompat library would handle this for us.

In the `Advanced Card Info` screen, the scan time is available from the card history list, which works properly when set to large size.  The card's raw UID is only shown in the history screen if the card is unknown – but there is limited utility showing this for _known_ cards.

If you're looking at writing transit data parsers and _really want it_, the JSON/XML exports have this as an attribute, and chances are you'll want to work with those files anyway (eg: loading binary data into a hex editor so that you get a nice display of all the binary data structures)... so I'm not concerned about this being unavailable.

In the `Map` screen, the start and end stations are also available in the trip history.  It's easy for these to be cut off anyway even at default font sizes with long station names (eg: Queensland bus stop names).

PR #745 (2021-01-02): Fix large font layout issues on Android
This fixes issues when Android is set to use the largest font and display sizes (fixes #744):

* 203a5f7289232a51dd39d01788320d63e08db990: always use multi-line labels for titles of preferences (which is also a problem for German at default font size)
* 482b0cb907bb5f92f027a03109d997709a0ed15e: wrap the About screen in a `ScrollView`
* 3bac09cd346e88e522b8b91ff7dd90c14b0dc268: move License screen's padding from the `ScrollView` to the `TextView`, so it has a larger viewport
* 6d35f303aa1fdfd74bca2acfc6789dd07f843c56: show full subscription names at large [... cut]
closed 2021-01-02 by micolous