metrodroid/metrodroid #754: [MFU] Québec OPUS card-based disposable tickets

austinhuang0131, 2021-06-07:
## New card checklist

- [x] I have read and acknowledged the New cards wiki page.
- [x] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [ ] I have attempted to understand the card format on my own.
- [x] I have provided notes on what I've found so far.

### Please answer all questions listed here:

- Name of the card: [L'Occasionnelle (STM Montréal)](http://stm.info/en/info/fares/opus-cards-and-other-fare-media/occasional-card), Occasionnelle ([RTC Québec City](https://www.rtcquebec.ca/en/rates-and-purchase/smart-cards/opus-card), STL Laval), [Solo Flex (RTL Longueuil)](https://www.rtl-longueuil.qc.ca/en-CA/fares/solo-flex-card/) and [Solo (exo Suburban Montréal)](https://exo.quebec/en/fares/opus-card/solo-card). (It is unclear whether STLévis has one.)
- Name of the transit agency: See above, all of which also participates in OPUS card.
- Transit agency's website: See above.
- Location(s): Across Québec, Canada. See above.
- Card media: MIFARE Ultralight

Is there any publicly accessible documentation for this card format?  If so, provide links:
No.

Is the card similar to one we already support? (if you don't know, that's OK)
They're treated like OPUS cards.

Is there an official app for using the card?
No, but [LecteurOPUS](https://github.com/etiennedub/LecteurOPUS), which is used by Metrodroid to use OPUS cards (see #126), fully supports them.

How many pages are on the card?
16.

Does the card appear as "fully locked" in Metrodroid?
No.

With the data that you can read from the card, does it change after performing a transaction?
Yes. It will deduct fares and write trips onto the card, which LecteurOPUS successfully reads.

phcoder, 2021-09-27:
Can you send me dumps I can test it on? You can find my email by adding @gmail.com to my nickname.

austinhuang0131, 2021-09-28:
Sent from username @ gmail.com.