metrodroid/metrodroid #758: [MFD] HOLO card (O‘ahu, Hawai‘i)

frozenpandaman, 2021-08-09:
Hi, this card recently (fully) rolled out and I'd love to help get it supported!

## New card checklist

- [x] I have read and acknowledged the New cards wiki page.
- [x] I can provide dumps, transaction history, and balance records for more than one card from the operator.
- [x] I have attempted to understand the card format on my own.
- [x] I have provided notes on what I've found so far.

## Please answer all questions listed here:

- Name of the card: HOLO
- Name of the transit agency: Oahu Transit Services, Inc.
- Transit agency's website: https://www.holocard.net/ & http://www.thebus.org/
- Location(s): O‘ahu (City and County of Honolulu), Hawai‘i
- Card media: MIFARE DESFire

Is there any publicly accessible documentation for this card format?  
No

Is the card similar to one we already support?  
Don't know, sorry!

Is there an official app for using the card?  
No

What application IDs are on the card?  
Not sure how to perform the procedure detailed [here](https://github.com/metrodroid/metrodroid/wiki/DESFire-Application-IDs) to check. Only relevant info seems to be Storage size = 16, Protocol = 5. Under raw data, file 0x0 is 96 bytes and file 0x1 is 48 bytes.

Does the card appear as "fully locked" in Metrodroid?  
No

With the data that you can read from the card, does it change after performing a transaction?
Can check next week and edit this post

frozenpandaman, 2021-10-24:
Got around to getting more info finally.

>With the data that you can read from the card, does it change after performing a transaction?

Yes. Card reads before and after a tap on TheBus are below:

[image]

[image]

ninjas28, 2021-11-09:
Hey @frozenpandaman, I've decided to try adding support for this card, but I don't have one of these cards to test with myself, would you be able to build the app and test it on your card (assuming you still have it)? I'm not quite ready to test it yet but I'm gonna try and work everything out in the next couple of days.

frozenpandaman, 2021-11-09:
@ninjas28 Wow, cool, thanks so much! This looks great so far. I actually just got a new computer a couple of days ago… one that's not almost a decade old, and *can* handle Android Studio on it. I'll download it and try building the app shortly. So yes, should be possible!

Just for fun, or as a souvenir of sorts, I could also mail you a card – I have some extras. Feel free to DM me on Twitter (linked in my bio) if you want. :)

frozenpandaman, 2021-11-11:
@ninjas28 Having some trouble building your fork/branch using Android SDK + Gradle from the command line (some Kotlin error – I really don't want to have to install Android Studio and deal with that all on this machine, tbh) but, re: your Twitter message and with it hopefully being ready to work on a real card, if you send me an .apk I'd be happy to test it out! But if not, no worries, you can test it too once your card arrives as well. :D

ninjas28, 2021-11-11:
@frozenpandaman I'll send you an APK tonight.

frozenpandaman, 2021-11-12:
@ninjas28 Awesome, it works great!

[image]
[image]

Amusingly it supports University of Hawai'i student ID cards too, which can be loaded with HOLO passes. Gets the number wrong there but that's expected, probably don't have to account for that. :)

Since the data's on the card, maybe it could support displaying the manufacturing number too? Especially since the whole ID number's not there.

ninjas28, 2021-11-12:
@frozenpandaman Yea I worked on adding the manufacturing ID late last night. I also need to change "Date of Issue" to "Date of last transaction" or some phrasing or other.

I'm also thinking of maybe just foregoing the first part of the serial number and just having it display the last 4 digits, since that's all that on the card and the question marks where the other numbers would be might confuse people who think that it was read incorrectly or corrupted or something. Might also just replace the question marks with asterisks.

frozenpandaman, 2021-11-14:
All sounds great to me! Not sure: re the incomplete serial number (I like the last four digits approach; if not, though, I agree `*`s are better than `?`s) but maybe one of the maintainers/authors can weight in at some point here or when you submit the PR!

closed 2022-06-16 by micolous