metrodroid/metrodroid #814: Orca: out of date balance and trips since new system roll out

darylharrison, 2022-08-06:
It looks like the transition to the new Orca system[1] has changed where balances and trips are stored on the card.

I have been using Metrodroid to keep track of my balance because it is simpler than using the official online account. However, since 2022-05-16, the balance has not been reducing (per Metrodroid), and only some of the trips have been recorded (and all of those said $0). I assumed this was due to issues related to the new system roll out until I boarded a bus and the Orca reader said "insufficient balance". I created an account on myorca.com and all my trips are recorded there and the balance is now negative.

[1] https://info.myorca.com/news/new-orca-system-for-regional-transit-launched/

ninjas28, 2022-08-10:
I've been tracking with a Proxmark3 the before and after dumps of my ORCA card and it doesn't look like the new readers change any data on the cards. The new readers in the system seem to communicate in real time to a backend server to update your balance and trip history online when you scan your card.

For the older readers that can't do real time updates, the ORCA people loaded a regional pass onto every card (you can actually see this in your card activity on myorca.com). This means that the old readers won't update your balance, since they don't think you're paying with an e-purse balance, so every trip will show as being 0.00 in Metrodroid. At the end of the day, or however often the old readers communicate with the backend system (remember the old 24-48 hour waiting period for online balance reloading?), the old readers send all the trips they processed to the backend system and it sorts out the fare payments and transfers and updates your account.
[image]

What this means for Metrodroid's ORCA reader going forward I'm not sure. It will probably end up being converted to a serial number only card reader, although we don't know what the format of the card data will be yet for the new black cards that are coming next month.

darylharrison, 2022-08-10:
Thank you for the information. This explains exactly what I've been observing. I'm curious how they will implement the balance display on card readers in the future because it can't be 100% reliable real time communication.

As for Metrodroid, I guess we'll wait and see what the end state looks like.

closed 2022-08-10 by darylharrison