metrodroid/metrodroid #818: [MFD] Support for new ORCA Card type

buckbanzai, 2022-09-28:
Hello, I have one of the newly re-designed ORCA cards (image of what the new card design looks like) that's beginning to roll out as part of the new system and have taken dumps of it after loading the card, and after completing a couple rides on it, and can provide that as well as screenshots of transaction history from the app. Not sure if there's anything useful that can be extracted, or if it's a moot point in light of #814.

## New card checklist

- [X] I have read and acknowledged the New cards wiki page.
- [ ] I can provide dumps, transaction history, and balance records for more than one card from the operator. - Only have the one new design card currently
- [X] I have attempted to understand the card format on my own.
- [X] I have provided notes on what I've found so far.

## Please answer all questions listed here:

- Name of the card: ORCA
- Name of the transit agency: Multiple
- Transit agency's website: myorca.com
- Location(s): Puget Sound, WA, USA
- Card media: MIFARE DESFire

Is there any publicly accessible documentation for this card format?  If so, provide links:
No

Is the card similar to one we already support? (if you don't know, that's OK)
New generation of the existing ORCA card, new black card design with updated logo and different card number format (mine is 19 digits, compared to the much shorter number formats on v1 ORCA)
[image]

Is there an official app for using the card?
iOS: https://apps.apple.com/us/app/myorca/id1606459618
Android: https://play.google.com/store/apps/details?id=com.myorca.myorca

What application IDs are on the card?
0x213f #0

Does the card appear as "fully locked" in Metrodroid?
No

With the data that you can read from the card, does it change after performing a transaction?
No

frozenpandaman, 2023-03-12:
This would be great to see – hoping it's supported soon!

hackerb9, 2023-05-02:
I can confirm that the new ORCA cards do not work. ORCA should probably be taken off the list of working cards.

Does anyone know if the problem with the new ORCA cards is that the data is now stored "in the cloud" or is it still on the card but hidden in some way?

lf-, 2024-09-21:
Can you extract the card number at least? That would be helpful: provide a link to check it online or something and prefill the card number without having to type the whole thing?

jakcron, 2025-08-31:
Well the new ORCA readers and Hop Fast Pass readers look to be the same model. So there probably are similarities between card formats.