metrodroid/metrodroid #819: [QUESTION] Policy on requesting keys from transit agency

kyuweftea, 2022-10-09:
I'm confused about whether or not this project supports automatically requesting keys from a transit agency in the app. On one hand, the wiki states:

> If you must query a remote web service for information about the card, then it will not be supported by this program.
[(link)](https://github.com/metrodroid/metrodroid/wiki/New-cards#looking-online-is-cheating)

On the other hand, I'm seeing support in the source code for requesting keys for TFI Leap card.

I'm asking since I'm in the process of reverse-engineering PRESTO card (Ontario, Canada). It contains mostly encrypted data, but it's readable by the official PRESTO app when online, similar to Leap. Only the serial number and some mystery bits are unencrypted. The balance and other information appear to be stored in the encrypted files.

Should I just submit a pull request with PRESTO as a serial-only card, or is it also worth trying to reverse-engineer the protocol used in the official app with the intent of adding support for reading the encrypted card data in metrodroid?

kyuweftea, 2022-10-10:
Nevermind, I found the answer to my question on the metrodroid main site:

> This project will only read data from the card itself, without having to connect to the agency’s back-office systems. In some cases, limited data is available, so balance information and trip history might not be available.

> Note: The Leap card reader connects to Transport for Ireland’s server for challenge-response authentication with the card. The data is otherwise interpreted locally, and connectivity is disabled by default.

closed 2022-10-10 by kyuweftea

micolous, 2023-05-21:
To answer this more concretely, a lot of that was written before Leap card support was a thing – and was actually written because of Myki. While I'm not familiar with PRESTO, I'll contrast and compare my thought process about Leap against two card which won't get carve-outs: Myki and Opal:

* Would using a web service for the card to get data make Metrodroid _essentially_ a glorified HTML screen scraper?

  * Leap: no: it sends us DESFire APDUs that let us unlock the card contents. Once unlocked, we can parse raw data.
  * Myki: yes
  * Opal: yes

  _Rationale:_ HTML screen scraping is very fragile. Using a website directly is generally better!

* If there is a web service to unlock the card, is that operated by the ticketing authority?

  * Leap: yes
  * Myki: n/a
  * Opal: n/a

  _Rationale:_ third-party data collection is (now) a barrier to app store reviews.

* Is there other useful data available _without_ the web service?

  * Leap: no
  * Myki: no
  * Opal: yes: it has 16 bytes of freely readable and useful data over NFC.

  _Rationale:_ there needs to be a strong reason to require a web service.

* Does the web service require that you create an account and register the card to use online functionality?

  * Leap: no
  * Myki: yes (and also asks for a lot of PII)
  * Opal: no, but without an account, you need to enter a 4 digit code printed on the back of the card, which we can't read over NFC

  _Rationale:_ not requiring accounts is good for data protection, and (now) is a barrier to app store reviews.

Any other card would at least need to meet the same criteria Leap does to get a carve-out.