metrodroid/metrodroid #829: [BUG] Bad PMm parsing?

dogtopus, 2023-03-10:
## Description of the issue

The PMm [maximum response time](https://github.com/metrodroid/metrodroid/blob/991bcd9a507558b51b9e9bcf12e04af24fea6db4/src/commonMain/kotlin/au/id/micolous/metrodroid/card/felica/FelicaCard.kt#L226-L243) calculation seems to be reading the bits backwards. In Figure 2-5 of the FeliCa manual it's clearly documented that e is bit 7:6, b is bit 5:3 and a is 2:0. However the code above seems to treat e as bit 1:0, b as bit 4:2, and a as bit 7:5.

Is there a particular reason to parse it this way?

## For card-specific issues (delete if not applicable)

- Name of the card type: Suica
- Name of the transit agency: JR
- Location: Japan

Please export a card dump from Metrodroid.

Please do not attach it to this issue -- you may be asked for it later via private email.

## Steps to reproduce

1. Scan any FeliCa.
2. Look at the Maximum Response Time section
3. The values do not match the ones calculated using formula documented in FeliCa manual section 2.3.4.

## Expected behaviour

Use algorithm described in FeliCa manual section 2.3.4 to calculate Max Response Time.

## Actual behaviour

The current Max Response Time algorithm does not match the one described in the manual, causing the resulting values to be wrong.

## Workarounds (if any)

None

## Device Logs

Android: Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

iOS: Generate a sysdiagnose using the instructions here: https://www.jessesquires.com/blog/how-to-sysdiagnose-ios/

Please do not attach it to this issue -- you may be asked for it later via private email.

## Details from About screen:

- Metrodroid version:
- Device Model:
- (Android only) Manufacturer:
- Android/iOS version:
- (Android only) NFC:
- (Android only) MIFARE Classic:

## Checklist

- [x] The card appears in the "supported cards" list in Metrodroid on my device.
- [x] I have copied all the details from the about screen into this bug.
- [ ] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [ ] I have generated an Android bug report ZIP, and can email it privately if needed.

dogtopus, 2023-03-11:
Seems that both NXP TagInfo and nfcpy agree with my interpretation and they seem to show the expected result.

micolous, 2023-05-21:
Yup, Metrodroid has the bits backwards.

[image]

commit 84db67b9 (2023-05-21): felica: implement calculateMaximumResponseTime with bits in the correct order (fixes #829)
closed 2023-05-21 by micolous