metrodroid/metrodroid #838: [BUG] Can't scan San Francisco Clipper cards

scottjg, 2023-08-14:
## Description of the issue

Physical Clipper cards do not appear to scan. I tried my card and my wife's card. It scans fine in NFC tools (granted, it provides no meaningful information). Card also works fine for actual transit.

Initially i thought it was maybe something iOS specific, but had the same problem on Android.

My card is pretty new, bought in the last 3mo.

Happy to help with any debugging information I can provide. I have an adafruit pn532 board that i can use as well if someone has specific thing they want me to try sending to the cards.

## For card-specific issues (delete if not applicable)

- Name of the card type: Clipper card (NFC tools says Desfire EV1 4k)
- Name of the transit agency: SFMTA
- Location: San Francisco

Please export a card dump from Metrodroid.

Please do not attach it to this issue -- you may be asked for it later via private email.

(i'm not sure how to do this since it doesn't scan)

## Steps to reproduce

1. Build the iOS app and deploy it to a device (in my case, an iPhone 12 Mini)
2. Click `Scan card`
3. Touch the clipper card to the phone

## Expected behaviour

it should scan the card

## Actual behaviour

it does not scan the card. just hangs on the scanning screen on ios:
[image]

on android, same deal.

## Workarounds (if any)

## Device Logs

Android: Generate a bug report ZIP using the instructions here: https://developer.android.com/studio/debug/bug-report.html

iOS: Generate a sysdiagnose using the instructions here: https://www.jessesquires.com/blog/how-to-sysdiagnose-ios/

Please do not attach it to this issue -- you may be asked for it later via private email.

## Details from About screen:

on android i ran the latest github release

- Metrodroid version: 3.0.0 (3920)
- Device Model: NE2215 (OP516FL1)
- (Android only) Manufacturer: OnePlus (OnePlus)
- Android/iOS version: 12 (SKQ1.211019.001)
- (Android only) NFC: On
- (Android only) MIFARE Classic: Supported

## Checklist

- [x ] The card appears in the "supported cards" list in Metrodroid on my device.
- [x ] I have copied all the details from the about screen into this bug.
- [ ] For card reading issues, I have exported a card dump from Metrodroid, and can email it privately if needed.
- [ ] I have generated an Android bug report ZIP, and can email it privately if needed.

scottjg, 2023-08-14:
on ios i also tried adding the generic desfire aid `d2760000850100` to the aid list, which surprisingly still didn't let the card scan.

scottjg, 2023-08-14:
also, the card scans fine in farebot on android.

hufman, 2024-03-09:
Good afternoon, fellow Clipper card user! I'm just a user of this app, but I've been hacking on it to fix some other bugs on my (older) Clipper card. Would you be able to send me a dump of your card and I can try taking a look at this crash?

scottjg, 2024-03-10:
the card doesn't scan in metrodroid so i don't think i can dump it from there. is there some other tool you want me to use?

hufman, 2024-03-10:
I quite like [TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite), which has a Full Scan tab and then can share the scan via email or any other means.